winder/Universal-G-Code-Sender #1405: UGS 2.0 doesn't launch (Fedora 32)

DynamiteStick, 2020-06-09:
## Problem description 
UGS 2.0 Doesn't launch on my fedora 32 machine. I'm fairly new to Linux so it could be I'm just inexperienced

### Steps
1. change directory to ugsplatform/bin
2. run the command ugspaltform

### Expected Behavior
UGS launches

### Actual Behavior
Nothing happens

### Specifications

#### Version

UGS Platform 2.0

#### Operating system

Linux Fedora 32

#### Platform

GRBL 1.1

breiler, 2020-06-10:
Are you using the latest nightly build with bundled Java from here?
https://github.com/winder/Universal-G-Code-Sender#downloads

Could you post the log file:
https://github.com/winder/Universal-G-Code-Sender/wiki/Configuration#log-files

cornnatron, 2020-06-11:
Its your run command lol
It should be ugsplatform not ugspaltform hihi

breiler, 2020-06-23:
Closing as the user hasn't responded

closed 2020-06-23 by breiler