winder/Universal-G-Code-Sender #1408: 2 Y-axis drivers?

cvhorie, 2020-06-15:
## Feature request

## Problem description 

### Expected Behavior

### Actual Behavior

### Steps to Reproduce the Problem

  1.
  1.
  1.

### Specifications
#### Version

#### Operating system

#### Platform

#### Other

cvhorie, 2020-06-15:
I am just starting with GRBL with a GRBLv1.1.board. This has 2 Y motor drivers, left and right, an X and no Z, attached to an eShapeoko with the 3 motors, 2*Y and X.  But the UGS Platform Wizard does not recognise the right Y driver and can test only the X and the left Y.  The Wizard needs to recognise the drivers present and then, if possible, the attached motors.

I apologise if this is the wrong way to report the problem, but there is no button to "raise a ticket". 
My PC is Win10 64 bit, fairly fast processor and good memory.  UGS Patform 2.0, latest version.

breiler, 2020-06-15:
Do you have links for your hardware where we can read the specification (I don't know what GRBLv1.1 board is)?
What are you hoping to achieve (control the two Y-axises separately or as one)?

cvhorie, 2020-06-15:
Thank you.
I mis-stated the board, It is a GAUPS 1.1, set to be compatible with gShield.
http://amberspyglass.co.uk/wiki/GAUPS_1.1_Instructions

The X-rail is 1.2 m long, so the two Y motors need to work in tandem - set to the same Y=0, oppositely rotating to keep the X-rail square with the Y-rails (0.75 m long).   Each Y motor/driver needs to be (?) separately tested and calibrated under load to achieve equal and reproducable movement across the X-rail.
I am at the beginning of my learning curve, so it is quite possible I have missed an option that provides for testing and tweaking of both Y motors.

breiler, 2020-06-15:
UGS don't know how many motors your hardware has connected and is up to your controller to handle. UGS only stream the gcode for the coordinates to step to and how to do that. The following would rapid move both Y-stepper motors to the absolute Y 10mm position:

```
G21 (use mm) 
G90 (absolute position) 
G0 (rapid movement 
Y10 (destination)
```

According to the manual of your controller there is an option to use dual-y motors which should be enabled by default: http://amberspyglass.co.uk/wiki/GAUPS_1.1_Instructions#Dual-Y_or_4-Axis_Operation

The best advice I can give you is to check your wiring again.

timon2jkl, 2020-06-16:
What he was saying the wizard only changes the value in GRBL in the first instance of the axis in question if you have 2 ti only changes on it the firmware

breiler, 2020-06-16:
@timon2jkl what do you mean? 

The wizard only sends g-code commands and there is no command for controlling separate motors on the same axis.

timon2jkl, 2020-06-17:
the wizard will change the value of $100-$102 but not $103 or $104

cvhorie, 2020-06-17:
Dear Breiler
You are better at diagnosis than I am. It was a dry solder joint in connecting the extension cable to the motor.  I apologise for not checking - chasing down the faults through many complexities, will get easier.  Thank you.

breiler, 2020-06-17:
@timon2jkl that would assume that @cvhorie is running a GRBL Mega or similar with support for more than three axises. 

I missed the comment from @cvhorie about:
> Each Y motor/driver needs to be (?) separately tested and calibrated under load to achieve equal and reproducable movement across the X-rail.

The options are as I see it:
  * Use a fourth axis as @timon2jkl mentioned like the one in [GRBL Mega](https://github.com/fra589/grbl-Mega-5X). But you won't get any visualizer support in UGS or any help from the setup wizard as it only supports three axises. I haven't seen many setups like this so I guess you will be threading uncharted territory here.
  * Use the hardware support on the GAUPS board that will clone the step and direction pulses to two stepper drivers. This will work with an "standard" GRBL setup and UGS, but with the downside that you can't calibrate each Y-axis separately.

@cvhorie did I understand you correctly that you managed to get the cloned second axis working using the hardware support on the GAUPS board?

timon2jkl, 2020-06-17:
I must have miss interpenetrated I thought he was using separate drivers like GRBL Mega not cloned I should have read closed to not muddy the waters. Glad he got it fixed.
Sorry

breiler, 2020-06-17:
@timon2jkl no worries, I wasn't aware of the extra configuration options in GRBL Mega. So you taught me something today. 😄

closed 2020-06-23 by breiler