winder/Universal-G-Code-Sender #1409: Add simple a simple vector graphics editor

breiler, 2020-06-16:
## Feature request
Universal G-code Sender has the primary focus to send existing gcode to the hardware. The CAD/CAM has to be done using other software. For beginners this may seem daunting to have to learn so many tools and techniques.

By adding support for doing simple tasks in Universal G-code Sender such as generating tool paths for cutting or engraving using images, vector graphics or to just engrave text we could lower the threshold for beginners to start making things.

This is a feature that I am currently working on and is keeping me up at nights. I'll try to log my work progress in this issue and hopefully get some feedback and insights from you guys.

Here is a small screen recording of the progress right now (still a lot to do). Open an SVG file and configure the cut operation on the geometric shape.
[image]

Please leave a comment of what you think?

khturnings, 2020-06-16:
This is looking awesome. 
Sometimes it is such a pain having to go to another program or in my case another computer when you just want to do something very simple like cut a circle out or a quick text engraving or just run a test on your machine.

Happy to test this for you if you need some testing.
Great work.

timon2jkl, 2020-08-06:
I like this very much I also like for simple things and testing very nice for the newbies to get them started and pointed in the right direction

krosspaper, 2021-06-26:
This is a fantastic feature, can't wait for it!

breiler, 2021-07-08:
I just want to give an update on the progress on this. There is still a lot of issues left to handle before it is ready for release, but I think it is coming along and I'm pleased with how it is turning out. It maybe doesn't look that different from the PoC screen cast posted earlier, but it is almost completely rewritten and I'm doing my best to making a solid base to continue working on.

In this demo screen cast I'm showing the basic functionality implemented so far and how it is currently integrated in UGS with NetBeans API:s:
* Creating a new design document
* Adding primitive geometries, performing basic transformations and setting the cut operation to pockets (moving and rotating)
* Importing an SVG 
* Changing the tool settings (tool diameter and "step over")
* Saving and Open

[image]

I will continue the work with tool path generation for different cut operations and remove the worst bugs. After this I'd appreciate some help with beta testing from you guys. 😀