winder/Universal-G-Code-Sender #1410: Joystick configuration issue?

Darren-Smithe, 2020-06-18:
Hi, loving the joystick implementation...all worked great on first try, but when selecting a new action to assign to a button, the list of available options seems strangely limited.

For example, there is the option to edit macros but not an option to run any predefined macros.

I wanted to use the letter labelled buttons to change the step size so that the dpad could be used for very fine tuning of position but the options do not seem to be present, is this a bug?

It would also be nice to prevent any actions unless another button is depressed (for example, the joysticks do not do anything unless a specific shoulder button is currently pressed). This would be ideal to prevent problems if the controller is dropped! I can understand if this is more of a feature request though :)

Thanks for such a great feature, I just wondered if the button configuration options were being incorrectly listed. Hopefully the image of the options uploads ok....
  
  UGS Platform 2.0 - Nightly Build Apr 25, 2020
  Windows 10
  GRBL 1.1f

[image]

breiler, 2020-06-22:
This doesn't seem correct. I can not reproduce it but it looks like it didn't discover all action categories.
I'll need to do some more testing.

The "dead man switch" is a good idea, I'll give it some thought.

Darren-Smithe, 2020-06-23:
Thanks Joacim. I’ve never dug into Java before so I thought I’d download the IDE and see if it was something Windows 10 specific.....no such luck as far as I can see!

The problem appears to be in the function recursiveAddCategoryActions.
Upon initial calling, the file object resolves (in my development setup) to:

C:\ugs\Universal-G-Code-Sender-master\ugs-platform\application\target\userdir\config\Actions

And this directory contains three subdirectories (Machine, Macro and Overrides).

When recursiveAddCategoryActions first calls getChildren(), I was expecting to see the three subdirectories mentioned, but instead it returns these three plus 13 other directories. I dug a little into getChildren() and it appears to be using an overlaid virtual file system...I don’t know if this is expected behaviour?!!?!

I’ll continue to have a look but not being a java programmer may lead me to incorrect conclusions :)

Darren-Smithe, 2020-06-23:
Working on the premise that even if there are a lot of unexpected directories in the list, the ones I would expect to see are in there as well....therefore it should be giving at least the options I’d expect. I had a look at the getActionFromFileObject function and that seems to be throwing quite a number of exceptions with the kind of functionality missing from the joystick config. I’ll attach the screenshot and leave it there as I’m not sure I’m helping! 
[image]

breiler, 2020-06-23:
Thank you for debugging this!

I don't have time looking into this right now, but these are good clues. =)

Darren-Smithe, 2020-06-24:
Hi Joacim, I believe I’ve found the answer although I don’t know the ‘correct’ way to fix it. 

The class ‘ActionRegistrationService’ creates an action reference cache called ‘actionCache’. This cache is created before a lot of the action classes have been instantiated, meaning they are omitted from the cache (the attempt to add them to the cache throws exceptions). The joystick configuration screen calls for category action information (function getCategoryActions) which returns the incomplete actionCache. I have proved this by changing getCategoryActions to reference a new cache which is only created on the first call to getCategoryActions. The menu options now look as expected. 

The above is clearly not a full fix as actionCache is used in another function of the class and this may be creating problems I have not seen. I don’t know enough about Java to either force class instantiation before actionCache is created or know enough about the app structure to move the actionCache creation to a better place.

I’ll test to see if my workaround fixes things for me in the short term and hope my java learning experience helps you or Will put a fix in place for us windows users.

Thanks again for the great joystick functionality!

All the best,
Darren

breiler, 2020-06-24:
Excellent work!

Have a look at this line: https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-platform/ugs-platform-ugslib/src/main/java/com/willwinder/ugs/nbp/lib/services/ActionRegistrationService.java#L45

This line will try to load the actions when the class is loaded as opposed to when the action service is used and instantiated. The class may very well be loaded before all actions. 

The initialization should be moved to a constructor which could solve the problem:
```
private Map<String, List<ActionReference>> actionCache;

public ActionRegistrationService() {
    actionCache = initCategoryActions();
}
```

Darren-Smithe, 2020-06-24:
Unfortunately, I have already tried this and it does not fix the problem :(

The cache building still throws ‘NoSuchMethodException’ for the majority of action classes resulting in many items missing from the cache.

PR #1413 (2020-06-25): Now invalidates the action cache if something is created or updated.
Fixes #1410 

Turns out I have the same problem once I fetched the latest code. I was working on another feature and thought I had the latest.

So I first tried using an annotation which will will run a method when starting (@OnStart) which didn't work. Then I tried one that will run once the GUI is showing (@OnShowing) which did work. But I realised that this would require a restart as soon as you add a macro or install plugins. 

So it will now invalidate the action cache if something is created or updated.
closed 2020-06-25 by winder