winder/Universal-G-Code-Sender #1411: Slow on Raspberry Pi 3 with bundled JDK

avarakin, 2020-06-23:
## Problem description 

I installed the latest nightly RPI installer with bundled Java 13 and it seems to be much slower than Java 8, which comes with Linux. Also, it seems that it consumes more memory.

### Specifications
#### Version
Latest nightly build

#### Operating system
  Ubuntu 20.04 32 bit on RaspberryPI 3 with Gnome DE

#### Platform
  GRBL 1.1

breiler, 2020-06-23:
Do you have any numbers and methods that made you come to this conclusion?

avarakin, 2020-06-24:
I am seeing startup time of about 30s for  Java 8 of Ubuntu 20.04 and about 55s for bundled Java 13. 
It was measured after starting/stopping 3 times in a row.

Utkarsh-vishnoi, 2020-07-19:
It would help if you tell us which model of PI are you using?
I use raspbian which is quite good on Raspberry Pi 3 Model B.
There can be a number of reasons for slow response time.

1. Your PI doesn't have enough empty memory.

breiler, 2021-01-08:
Downgraded to jdk8 by default in the rpi-package.

closed 2021-01-08 by breiler