winder/Universal-G-Code-Sender #1412: Joistick is not working

avarakin, 2020-06-23:
## Problem description 
Most likely I am missing something basic, but I can not get Joystick working. 
I enabled the plugin in the plugins screen but when I come to the settings screen, which has the configuration for Joystick, I am not seeing any settings and checkbox for enabling Joystick. 
I was able to get it working on my PC few weeks ago but for some reason, it does not work on my Pi3.

### Specifications
#### Version
Latest nightly for Pi

#### Operating system
  Ubuntu on RaspberryPI 3

#### Platform
  GRBL 1.1f

breiler, 2020-06-23:
A screenshot of the settings screen would help.

Also if you Could post your log file: https://github.com/winder/Universal-G-Code-Sender/wiki/Configuration#log-files

avarakin, 2020-06-24:
I think I found the relevant error in the log:

INFO [com.willwinder.ugs.nbp.core.lifecycle.startup]: Loading JogService...
SEVERE [org.openide.util.RequestProcessor]: Error in RequestProcessor com.willwinder.ugs.nbp.joystick.Startup
java.lang.UnsatisfiedLinkError: /tmp/jnigen/923140537/libjamepadArm.so: libbcm_host.so: cannot open shared object file: No such file or directory
        at java.lang.ClassLoader$NativeLibrary.load(Native Method)
        at java.lang.ClassLoader.loadLibrary0(ClassLoader.java:1934)
        at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1817)
        at java.lang.Runtime.load0(Runtime.java:809)
        at java.lang.System.load(System.java:1088)
        at com.badlogic.gdx.jnigen.JniGenSharedLibraryLoader.loadLibrary(JniGenSharedLibraryLoader.java:105)
        at com.badlogic.gdx.jnigen.JniGenSharedLibraryLoader.load(JniGenSharedLibraryLoader.java:191)
        at com.studiohartman.jamepad.ControllerManager.<init>(ControllerManager.java:65)
        at com.studiohartman.jamepad.ControllerManager.<init>(ControllerManager.java:42)
        at com.willwinder.ugs.nbp.joystick.service.JoystickServiceImpl.<init>(JoystickServiceImpl.java:93)
        at com.willwinder.ugs.nbp.joystick.Startup.run(Startup.java:50)
        at org.openide.util.RequestProcessor$Task.run(RequestProcessor.java:1418)
        at org.netbeans.modules.openide.util.GlobalLookup.execute(GlobalLookup.java:45)
        at org.openide.util.lookup.Lookups.executeWith(Lookups.java:278)
[catch] at org.openide.util.RequestProcessor$Processor.run(RequestProcessor.java:2033)

breiler, 2020-06-24:
Aha, it looks like it is depending on the /opt/vc directory. And this directory needs to be removed/moved for the visualizer to work...

khturnings, 2020-06-24:
At the end of issue https://github.com/winder/Universal-G-Code-Sender/issues/1247 it is noted that you can use a script to run UGS that moves the /opt/vc during startup and then in the last post it mentions that it can be reinstated after a pause.

Its a bit of a workaround but might help.

avarakin, 2020-06-28:
My original issue was caused due to ubuntu missing VC.
I migrated to Rasbpian and now I have catch 22 situation - rename vc, joystick does not work, restore vc, visualizer does not work.

This post on jogl forum suggests that version of jogl library is outdated and the issue with visualizer can be fixed by using the latest builds of jogl:
http://forum.jogamp.org/Available-GLVersions-not-set-for-EGLGraphicsDevic-td4040111.html

breiler, 2020-06-30:
We are actually using the latest build 2.3.2 and the fix will be available in 2.4.0. We will just have to wait. 😞

breiler, 2023-04-25:
Closing as we can't reproduce the error and we didn't get a response from the user.

closed 2023-04-25 by breiler