winder/Universal-G-Code-Sender #1415: Unable to Open G-Code File

tfjield, 2020-06-24:
## Problem description 
When I launch UGS Platform it starts fine, but if I try to open a g-code file the open file dialog is never shown.

### Expected Behavior
The open file dialog should be shown so that I can choose a file.

### Actual Behavior
The UI becomes partially frozen.  I can move the window, but I can't access any other menu items.  I can't close it either; I have to kill it with Task Manager.

### Steps to Reproduce the Problem
1.  Open the program.
2.  Click the folder icon to open a file.
3.  Wait...  for eternity.

### Specifications
#### Version
UGS Platform Version 2.0 [nightly] / Apr 25, 2020
Product Version: Universal Gcode Platform 20200425
Java: 1.8.0_251; Java HotSpot(TM) Client VM 25.251-b08
Runtime: Java(TM) SE Runtime Environment 1.8.0_251-b08
System: Windows 10 version 10.0 running on x86; Cp1252; en_US (ugsplatform)

Both 32 and 64 bit behave the same.

#### Operating system
Windows 10.

#### Platform
GRBL 1.1f, but haven't gotten that far.

#### Other
N/A

tfjield, 2020-06-24:
More info:  I left the program running in the background and the open file dialog finally popped up around 6 to 8 minutes later.  If I try to open another file, it sleeps again...

dwaal, 2020-06-28:
Is this info missing..?

How did you create this G-Code file? With what software?
Did you attach the G-Code file? ...So we can have a look at it.

P.S.: to devs...is there a list of recommended software that outputs CNC g-code and has zero problems with UGS?

breiler, 2020-06-29:
@tfjield this could be a problem with your Java, could you post your log file: https://github.com/winder/Universal-G-Code-Sender/wiki/Configuration#log-files

You could try the windows package that we are providing here: https://github.com/winder/Universal-G-Code-Sender#downloads
It comes with a bundled Java which means that you don't need to install it seperatly.

@dwaal what CAM programs that is suited depends on what controller you are using (GRBL and its different forks, TinyG, G2core and Smoothieware). I have added a section for CAM softwares in the software comparison page here, complement it with the softwares you have experience with: https://github.com/winder/Universal-G-Code-Sender/wiki/Software-comparison

eGit, 2020-06-29:
@breiler Interesting section you added. Isn't Kiri:Moto is not only free to use but MIT licence: https://github.com/GridSpace/grid-apps

Also it should understand a lot of controllers (empty in your table cell)

Just started using UGS. Nice tool :)
Is there no way to simulate the toolpath (without connecting to a GRBL controller)?

breiler, 2020-06-29:
@eGit thanks, I've updated the license. I haven't used it yet so I can't vouch for it. Just stumbled upon it yesterday and wanted to add it as it looks very promising.

tfjield, 2020-06-29:
@dwaal, are you speaking to the issue I posted?  It's not an issue with the g-code file, because the problem occurs before I select a file.  The problem is that the program freezes before the open file dialog appears.

@breiler, I've looked for the log file, but there is no directory called ugsplatform under Roaming.  I searched AppData and there isn't a folder called ugsplatform anywhere.

I didn't have Java installed before this, so it's a fresh install.  I installed Java 8 Update 251 both in 32 and 64 bit.

I just downloaded the Windows version from the link you sent (I had previously downloaded from the downloads page) and ran the executable.  It said to "Unpack and start the program bin/ugsplatform" and there is a script file in that directory without an extension, but I'm assuming I should run one of the executables, 32 or 64 bit?  It behaves the same, and has the same problem.

I just noticed that config files are stored in C:/users/[your username>]/.ugsplatform, so I checked there and I was able to find a log file here:  C:\Users\Todd Fjield\.ugsplatform\2.0-SNAPSHOT\dev\var\log, not under AppData.  "messages (2020-06-24).log" is from the day I posted this issue.  "messages.log" is from the windows version today, which I grabbed right after the open file dialog became visible.

[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/4847362/messages.log)
[messages (2020-06-24).log](https://github.com/winder/Universal-G-Code-Sender/files/4847350/messages.2020-06-24.log)

Thanks for your help!

tfjield, 2020-06-29:
@breiler, just to be clear, I tried the -win version before and after uninstalling Java.  Same result in both cases, so it does look like it's using the included JRE.  I've also disabled all antivirus and firewalls.

The CPU usage sits at 0% when it's waiting for the open file dialog to appear.

I've attached the updated log file using the -win version with Java uninstalled, in case this helps narrow the problem down.

[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/4848062/messages.log)

breiler, 2020-06-30:
The log files didn't reveal anything and I currently don't have access to a Windows machine. So I have difficulties to help. 😞 
If anyone else have a similar problem I'd be interested to know.

tfjield, 2020-06-30:
So I made some progress debugging this.  I'm a coder, just not Java.  But I installed NetBeans and downloaded the source so I could figure out where the delay was.  Before I even got that far, I had the same problem in NetBeans; the file open dialog was taking forever to show.

I did some searching online and found this [https://bugs.java.com/bugdatabase/view_bug.do?bug_id=6372808](url) and this [https://bugs.java.com/bugdatabase/view_bug.do?bug_id=6584600](url). 

By default, the file open dialog was trying to open my Documents folder, in which I had a couple zip files.  Not extremely large; just 50 megs each or so.  Moving these files to a subdirectory has effectively fixed the problem.  The open file dialog now appears with a delay of only 10 seconds or so, which is usable.  The 10 seconds is probably due to the number of folders in Documents.

So it looks like this is a Java issue that's been around forever, and not a problem with UGS.

Thanks for your help!

closed 2020-07-17 by breiler