winder/Universal-G-Code-Sender #1417: Classic (4/25/2020) won't jog

loonster, 2020-06-28:
## Problem description 
Running Linux Ubuntu 20.04 and recommended Oracle java (built _251).  Classic UGS loads OK, homes OK but will not jog in x,y, or z axes.

### Expected Behavior
Should jog

### Actual Behavior

### Steps to Reproduce the Problem

  1. Open the program, connect to GRBL, home.
  2. Hit button (or key equivalent) to jog.
  3. Stepper motor grinds but no motion.

### Specifications
#### Version
UGS Classic 4/25/2020 build

#### Operating system
Linux Ubuntu  20.04

#### Platform
  GRBL 1.1f#### Other
No unusual messages from terminal.

breiler, 2020-06-28:
Do you get any motion at all besides homing? I.e if you run a gcode program or commands manually?

Could you post your GRBL settings. Should be displayed right after connecting.

loonster, 2020-06-28:
Hah!  After posting yesterday, I ran the Ubuntu upgrade utility, which downloaded 250MB of patches, etc., and this morning everything works OK.  Can jog,  issue commands manually, and run gcode programs.  Solved.   Thanks for the quick response!.
 
Tim

closed 2020-06-28 by loonster