winder/Universal-G-Code-Sender #1423: Bluetooth Port (/dev/rfcomm0) not showing up in the ports list

Utkarsh-vishnoi, 2020-07-19:
## Problem description 
The application can only detect the wired serial communication port but not the bluetooth serial communication port.
In my case it's /dev/rfcomm0. I have modified my CNC setup to work over bluetooth which is working like a charm from arduino serial console. but with the UGS the rfcomm0 port isn't showing up. 

### Expected Behavior
When I try to refresh the ports list, it doesn't show the bluetooth serial port (/dev/rfcomm0) but the port is visible in the arduino serial console. Also when I send `$$` in serial console, I get the expected output. It will be really helpful to just communicate over the bluetooth serial port as well.

### Actual Behavior
The bluetooth serial port is not detected by the application

### Specifications
#### Version
UGS Platform 2.0 - Nightly Build Aug 14, 2019

#### Operating system
Raspbian on the Raspberry PI

#### Platform

GRBL 1.1f on Arduino Uno (Works completely fine on wired serial communication)

closed 2020-07-28 by Utkarsh-vishnoi

breiler, 2020-07-28:
Did you get it working?

Utkarsh-vishnoi, 2020-07-30:
Yes, I was always using dropdown for the ports. If I enter the port name manually, I was able to get it to work.