winder/Universal-G-Code-Sender #1424: Problem in the setup wizard

Raini20, 2020-07-21:
## Problem description 
Setup Wizard strange behaviour at defining the steps/mm and impractical order of limit switch setup 

### Expected Behavior
When I use the setup wizard and try to define the steps/mm I have to click UPDATE after every change. The actual position should not change.

### Actual Behavior
After clicking UPDATE the position changes to a (random?) value and I can not set it to zero. A click on X+ or X- suddenly makes a movement from the strange value to zero. It should drive 1mm. Since there are no limit switches configured yet it made a serious damage to my machine

### Solution
#### A) fixing the bug
#### B) changing the order of limit switches and steps/mm in the setup wizard

### Specifications
#### Version  
  UGS Platform 2.0 - Nightly Build Jul 17, 2019

#### Operating system

  Windows 10

#### Platform

  GRBL 1.1h

breiler, 2020-07-21:
I'm sorry to hear it caused damage to your machine! 😞 

I would really recommend the latest nightly build (https://github.com/winder/Universal-G-Code-Sender#downloads), but in this case it wouldn't matter as the bug exists in both versions.

Thanks for the report!

henfri, 2020-11-09:
>  am just guessing here, this could happen if the internal state in GRBL is keeping track of the number of steps it has taken and not the actual position. If the step size configuration changes the number of steps would be recalculated to a new position

I had the same impression.
Thus I suggest the workaround mentioned in theothher issue (reset actual position to 0mm after klicking "update"

breiler, 2020-11-09:
Yes that should be the way to do it.

We can only reset the position after the command has been processed, the communication is asyncronous so this becomes a bit tricky.

breiler, 2020-11-09:
It could be a problem with GRBL, I remember encountering it when I first started with my CNC. 

I can only resolve this with a soft reset after changing the setting which is a bit violent. I've posted a bug report over at GRBL and we'll see if they have any ideas: https://github.com/gnea/grbl/issues/955

henfri, 2020-11-11:
Hello,

I understand, thanks.
In case it is not fixes in grbl, I think that a reset is ok. Alternatively, UGS could use its own counter for x/y/z during the wizard.

Regards,
Hendrik

PR #1489 (2020-12-08): Fixed problem with step size settings being loaded properly.
Now resets the controller when updating the step size configuration to force the controller to load the new settings.
Fixes #1424
closed 2020-12-08 by breiler