winder/Universal-G-Code-Sender #1425: Gamepad not recognized

JensHauser, 2020-07-26:
## Problem description 
I have a new gamepad, which is installed in Windows 10 and I´m able to calibrate it and confirm that all buttons/sliders work in the device manager. I activated the gamepad in UGS. 

### Expected Behavior
As written in the docs, pressing a button should flash the corresponding activity in UGS

### Actual Behavior
No clicks are recognized, it feels like there was no gamepad.

### Specifications
#### Version
UGS Platform 2.0 - Nightly Build July 17, 2020

#### Operating system
Windows 10, all updates installed

#### Platform
GRBL 1.1f

breiler, 2020-07-26:
What type of gamepad is it?

JensHauser, 2020-07-26:
„EasySMX Gaming Controller, 2.4G Wireless Gamepad, PS3 Controller, Dual Vibration“

 

https://www.amazon.de/gp/product/B07D6P48J9/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8 <https://www.amazon.de/gp/product/B07D6P48J9/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1> &psc=1

breiler, 2020-07-26:
This is a list of controllers that will be recognised by the gamepad library we are using:
https://github.com/electronstudio/sdl2gdx/blob/master/res/gamecontrollerdb.txt

I couldn't find it in that list, could you post your log file:
https://github.com/winder/Universal-G-Code-Sender/wiki/Configuration#log-files

JensHauser, 2020-07-26:
A fresh log file.
I started UGS, went to Tools menue and tried to tick the check box "Active joystick".
Then the cross disappeared. Tried several times.

[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/4977183/messages.log)

breiler, 2020-07-26:
The logs didn't reveal anything. I've created a new version with additional logging to see if it detects the gamepad controller. and I would appreciate if you could download the latest nightly build and try again and post the log file.

JensHauser, 2020-07-26:
Thank you for taking care about this.
I installed the new nightly build and here are 2 log files and some screenshots that show the game pad running under Windows 10.

[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/4977283/messages.log)
[messages2.log](https://github.com/winder/Universal-G-Code-Sender/files/4977284/messages2.log)
[Dok1.pdf](https://github.com/winder/Universal-G-Code-Sender/files/4977285/Dok1.pdf)

JensHauser, 2020-07-26:
Some infos producers [website](https://www.easysmx.com/collections/game-controllers/products/easysmx-kc-8236-wireless-gamepad)

breiler, 2020-07-26:
As I suspected. The library we are using can't identify your controller. But there is still hope. 
The configuration file is a bit outdated and a newer one is provided in SDL (which is used by the library we are using).  https://github.com/gabomdq/SDL_GameControllerDB/blob/master/gamecontrollerdb.txt

I'll investigate if we can get the database file updated in some way.

breiler, 2020-07-26:
It's a long shot and there was no way for me to test this. I've updated the db-file which is loaded by the library. Try the new version now.

JensHauser, 2020-07-26:
Works out of the box :-)
Thanks a lot for your quick and comprehensive support

closed 2020-07-26 by JensHauser