winder/Universal-G-Code-Sender #1426: Macro and calculation

JensHauser, 2020-07-26:
Is it possible to add a macro and calculate a number based on current machine coordinates?

G43.1 Z{machine_z-1} or so would be helpful.

If not, do you have an idea how to come closer to this functionality?

JensHauser, 2020-07-29:
Is this functionality possible by writing a plug-in?

breiler, 2020-07-31:
Thanks for the suggestion and everything is possible! 😁 

This should be possible to implement as a plugin but might as well be implemented as a core functionality. This should probably be added using the [processor API](https://github.com/winder/Universal-G-Code-Sender/tree/master/ugs-core/src/com/willwinder/universalgcodesender/gcode/processors).

We have made a similar thing in the DRO allowing the use of expressions to calculate work positions using a JavaScript environment in [GUIBackend](https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-core/src/com/willwinder/universalgcodesender/model/GUIBackend.java#L772):

[image]

If we define the variables `machine_x`, `machine_y`, `machine_z`, `work_x`, `work_y`, `work_z` in the script environment we could use the `{ }` as a general tag for parsing JavaScript and making it possible to use all kinds of arithmetics and logic. 

We could then remove the `{ }` processing from [MacroHelper](https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-core/src/com/willwinder/universalgcodesender/MacroHelper.java) as this would be more general.

One thing that we would need to solve is the `{prompt|name}` function and needs to be investigated further if this could be executed from within the script environment.

JensHauser, 2020-07-31:
And in case g code needs to be executed and we expect a result, e.g. current position after G38.2, it should wait for finished movement and make the result usable for calculating.

breiler, 2022-02-02:
For future reference, this is how cncjs does it:
https://github.com/cncjs/cncjs/blob/b1a82928294b6c82f63077e2bfee856c857636ce/src/server/lib/evaluate-assignment-expression.js

julien8745, 2024-02-13:
Good morning,
I want to control an arm with two Z links via an arduino and GLBR. But GLBR is only programmed for linear axes and not angular connections. I would therefore like to make a macro which retrieves the coordinates of the points to be reached in the gcode, converting them via the inverse geometric model of my robot into angles. is it possible to do this via a macro in CNCJS?

breiler, 2024-02-13:
@julien8745 please create a new discussion. 

If you have questions regarding CNCJS I would recommend asking it on their project page: https://github.com/cncjs/cncjs

The coordinate can be in whatever unit you want. GRBL will report it in either metric or imperial units. But you can just as well treat it as degrees if you want to. 

There is no macro function in UGS to do calculations. If it helps, you can record positions using this feature: https://github.com/winder/Universal-G-Code-Sender/pull/2307