winder/Universal-G-Code-Sender #1428: UGS does not send cntl-X to GRBL.

J-Dunn, 2020-07-26:
## Feature request
New button to send cntl-X to GRBL

## Problem description 
If I want to stop a job and clear the command buffer in GRBL ( without doing a reset which loses machine position ) I need to send cntl-X character to GRBL.

This does not seem possible from UGS.  It seems the UGS java UI swallows a cntl-X typed from the keyboard. This means we need a button or other UI widget to do this explicitly. 

Currently the best I can do tell UGS to stop sending and wait for GRBL to finish executing all commands already sent and hope that nothing bad happens in the meantime. 

If I need to stop a job this very likely means I need to stop it NOW, not later. It seems that the only way out currently is to reset GRBL which implies recovering and resetting machine position which wastes a lot of time and  inevitably introduces errors in the job. 

### Expected Behavior
Pause works well and instantly stops the machine movement. What is needed is a means to get out of that state without doing a resume, then killing further sending and waiting for all the sent commands to run their course. 

### Steps to Reproduce the Problem
1. send job to GRBL
2. see impending urgency ( eg. impending tool crash ) 
3. press "Pause" button
4. now what ? ..... 

### Specifications
#### Version

  UGS Classic 1.0.9

#### Operating system
Linux 

#### Platform
GRBL

breiler, 2020-07-28:
That's an old issue in UGS Classic 1.0.9, try the latest nightly build. In the newer version you are able to press "Stop" after everything is paused.

Download the newest version from here:
https://github.com/winder/Universal-G-Code-Sender#downloads

J-Dunn, 2020-07-28:
Many thanks for the reply.  I was looking at the newer versions of platform and classic. Is there a way to send Ctrl-X in classic v2? 

 It seems I can define a macro in platform too. I was going to look into that.

It seems that you are referring to the play, pause , stop icons in platform. The functionality is there which is great, but I find these features too small for machine control in the practical context of machine control in the workshop.

When I need to stop a job it is often because I see something going badly wrong but short of real emergency where I need to hit the E-stop on the machine, lose machine position etc.  However, I do need to get to this FAST. 

On classic, I position the mouse over the pause button as soon as I set the job running. That way I can just tap the scroll pad to put the job on hold. I still need to be careful not to drift the mouse position as I tap, that slows me down little when I need to move as fast as possible. The target needs to be as large as possible to ensure that some small movement does not drift off the button. The much smaller icon is not as good in that sense. 

Ideally I would like to be able to set the keyboard focus on some input element and then I can tap the spacebar to activate it.  That is something requiring less precision than an on-screen reaction, which I can thus do quicker in a rush. 

The video recorder icon paradigm is fine in general but I need something large and accessible in this context. 

All the extra functionality which UGS now has is great but it seems like the core function of controlling the sending a job to the controller is somehow being relegated and losing _necessary_ accessibility. 
 
Other panels can be moved and resized, but this iconified menu is not modifiable. Maybe the recorder buttons could be in a dockable window which can be resized _front and centre_ while actually running a job ? 

Something like the "common actions" tab could easily include Run Pause Stop as large buttons with pause being largest and getting the keyboard focus after Run is clicked. 

That would probably require minimal programming effort to implement. 

I hope these suggestions are useful. Thanks for your reply.

winder, 2020-07-28:
When the machine is paused you can stop it as a follow-up step.

One thing that might help would be adding a keybinding for the pause button
so that you don't need to worry about positioning the mouse every time.

There is also a joystick mode now, which you could use as a more reliable
physical input.

Thank you for the feedback. A larger and more prominent emergency stop
sounds like a good safety measure that would improve UGS. Where do you
think the best place for this button would be? A new window would be the
normal way we add these sorts of extensions.

breiler, 2020-07-28:
If you use the latest version:
* The pause button will send a `!` real time command to GRBL which will pause almost immediately (if in motion, the machine will safely decelerate to a stop and then be suspended). It's then either possible to stop or resume the program.
* The stop button will send a `!` real time command, clear any queued commands then send a 0x18 (CTRL-X) to clear the command buffer and reset GRBL. It will retain the position as the controller is not in motion due to the previous pause.

So the feature that you describe is in place. There is no "urgent pause" in GRBL, if you simply send the CTRL-X without checking if GRBL is still in motion will cause the loss of position and probably a ruined material.

@winder and I have previously discussed a configurable panel for different actions which should now be possible. A new custom window with Run, Pause, Stop actions as you described should also be easy to implement.

Good stuff, thanks for the feedback!

J-Dunn, 2020-07-28:
Thanks for the detail on what the Stop icon does. Sending ! first is probably a good idea. 

I recognised that all the functionality was there is earlier replies, it is just the accessibility. Since the idea has been warmly received, I'm sure you all come up with a great implementation. 

Thanks for being responsive.

breiler, 2020-10-28:
Closing as it is now possible to customize your own actions-panel: #1458

closed 2020-10-28 by breiler