winder/Universal-G-Code-Sender #1431: Autoleveler Apply to Gcode error: Internal parser error: missing data.

agenteDserrano, 2020-07-29:
### Expected Behavior
Apply to Gcode after load file, generate test data or Scan surface.

### Actual Behavior
When I try to apply to Gcode after load file, generate test data or Scan surface I get the error: Internal parser error: missing data.

### Steps to Reproduce the Problem

[image]

>>> $$
$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 6    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 3    (Status report options, mask)
$11 = 1.000    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 0    (Homing cycle enable, boolean)
$23 = 7    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 2.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 800.000    (X-axis travel resolution, step/mm)
$101 = 800.000    (Y-axis travel resolution, step/mm)
$102 = 800.000    (Z-axis travel resolution, step/mm)
$110 = 2000.000    (X-axis maximum rate, mm/min)
$111 = 2000.000    (Y-axis maximum rate, mm/min)
$112 = 100.000    (Z-axis maximum rate, mm/min)
$120 = 20.000    (X-axis acceleration, mm/sec^2)
$121 = 20.000    (Y-axis acceleration, mm/sec^2)
$122 = 20.000    (Z-axis acceleration, mm/sec^2)
$130 = 500.000    (X-axis maximum travel, millimeters)
$131 = 500.000    (Y-axis maximum travel, millimeters)
$132 = 200.000    (Z-axis maximum travel, millimeters)
ok

AutoLeveler data:
[
  {
    "x": 0.0,
    "y": 0.0,
    "z": -0.21
  },
  {
    "x": 0.0,
    "y": 10.0,
    "z": -0.038
  },
  {
    "x": 0.0,
    "y": 20.0,
    "z": -0.069
  },
  {
    "x": 0.0,
    "y": 30.0,
    "z": -0.086
  },
  {
    "x": 0.0,
    "y": 40.0,
    "z": -0.098
  },
  {
    "x": 0.0,
    "y": 48.164,
    "z": -0.081
  },
  {
    "x": 10.0,
    "y": 0.0,
    "z": 0.029
  },
  {
    "x": 10.0,
    "y": 10.0,
    "z": -0.029
  },
  {
    "x": 10.0,
    "y": 20.0,
    "z": -0.076
  },
  {
    "x": 10.0,
    "y": 30.0,
    "z": -0.105
  },
  {
    "x": 10.0,
    "y": 40.0,
    "z": -0.114
  },
  {
    "x": 10.0,
    "y": 48.164,
    "z": -0.099
  },
  {
    "x": 20.0,
    "y": 0.0,
    "z": 0.079
  },
  {
    "x": 20.0,
    "y": 10.0,
    "z": -0.019
  },
  {
    "x": 20.0,
    "y": 20.0,
    "z": -0.076
  },
  {
    "x": 20.0,
    "y": 30.0,
    "z": -0.101
  },
  {
    "x": 20.0,
    "y": 40.0,
    "z": -0.103
  },
  {
    "x": 20.0,
    "y": 48.164,
    "z": -0.088
  },
  {
    "x": 30.0,
    "y": 0.0,
    "z": 1.763
  },
  {
    "x": 30.0,
    "y": 10.0,
    "z": 0.011
  },
  {
    "x": 30.0,
    "y": 20.0,
    "z": -0.056
  },
  {
    "x": 30.0,
    "y": 30.0,
    "z": -0.054
  },
  {
    "x": 30.0,
    "y": 40.0,
    "z": -0.088
  },
  {
    "x": 30.0,
    "y": 48.164,
    "z": -0.078
  },
  {
    "x": 40.0,
    "y": 0.0,
    "z": 0.125
  },
  {
    "x": 40.0,
    "y": 10.0,
    "z": 0.013
  },
  {
    "x": 40.0,
    "y": 20.0,
    "z": -0.04
  },
  {
    "x": 40.0,
    "y": 30.0,
    "z": -0.075
  },
  {
    "x": 40.0,
    "y": 40.0,
    "z": -0.08
  },
  {
    "x": 40.0,
    "y": 48.164,
    "z": -0.06
  },
  {
    "x": 50.0,
    "y": 0.0,
    "z": 0.135
  },
  {
    "x": 50.0,
    "y": 10.0,
    "z": 0.023
  },
  {
    "x": 50.0,
    "y": 20.0,
    "z": -0.035
  },
  {
    "x": 50.0,
    "y": 30.0,
    "z": -0.056
  },
  {
    "x": 50.0,
    "y": 40.0,
    "z": -0.065
  },
  {
    "x": 50.0,
    "y": 48.164,
    "z": -0.049
  },
  {
    "x": 60.0,
    "y": 0.0,
    "z": 0.155
  },
  {
    "x": 60.0,
    "y": 10.0,
    "z": 0.043
  },
  {
    "x": 60.0,
    "y": 20.0,
    "z": -0.014
  },
  {
    "x": 60.0,
    "y": 30.0,
    "z": -0.048
  },
  {
    "x": 60.0,
    "y": 40.0,
    "z": -0.059
  },
  {
    "x": 60.0,
    "y": 48.164,
    "z": -0.045
  },
  {
    "x": 70.0,
    "y": 0.0,
    "z": 0.188
  },
  {
    "x": 70.0,
    "y": 10.0,
    "z": 0.079
  },
  {
    "x": 70.0,
    "y": 20.0,
    "z": 0.009
  },
  {
    "x": 70.0,
    "y": 30.0,
    "z": -0.03
  },
  {
    "x": 70.0,
    "y": 40.0,
    "z": -0.035
  },
  {
    "x": 70.0,
    "y": 48.164,
    "z": -0.02
  },
  {
    "x": 80.0,
    "y": 0.0,
    "z": 0.243
  },
  {
    "x": 80.0,
    "y": 10.0,
    "z": 0.12
  },
  {
    "x": 80.0,
    "y": 20.0,
    "z": 0.053
  },
  {
    "x": 80.0,
    "y": 30.0,
    "z": 0.014
  },
  {
    "x": 80.0,
    "y": 40.0,
    "z": 0.003
  },
  {
    "x": 80.0,
    "y": 48.164,
    "z": 0.136
  },
  {
    "x": 90.0,
    "y": 0.0,
    "z": 0.266
  },
  {
    "x": 90.0,
    "y": 10.0,
    "z": 0.17
  },
  {
    "x": 90.0,
    "y": 20.0,
    "z": 0.105
  },
  {
    "x": 90.0,
    "y": 30.0,
    "z": 0.071
  },
  {
    "x": 90.0,
    "y": 40.0,
    "z": 0.061
  },
  {
    "x": 90.0,
    "y": 48.164,
    "z": 0.071
  },
  {
    "x": 94.968,
    "y": 0.0,
    "z": 0.314
  },
  {
    "x": 94.968,
    "y": 10.0,
    "z": 0.194
  },
  {
    "x": 94.968,
    "y": 20.0,
    "z": 0.132
  },
  {
    "x": 94.968,
    "y": 30.0,
    "z": 0.103
  },
  {
    "x": 94.968,
    "y": 40.0,
    "z": 0.093
  },
  {
    "x": 94.968,
    "y": 48.164,
    "z": 0.099
  }
]

mbidaki76, 2020-09-25:
I have same problem

justmyhobby, 2020-11-18:
For me, this error happens at line to set feed rate (G01 F....) without X, Y or Z. For example: G01 F100. 
To fix this, just remove the code G01 and leave the F.... alone and then run "Apply to GCode" again.
BTW, I'm using GRBL driver.

Franz-Juergen, 2021-03-04:
Many thanks for the great tool - I just run into the same issue as several times reported. Thank You to <justmyhobby> for the hint. With this I was able to solve my issue. - I just replaced all strings "G01 F" with "F" within the relevant gcode file.

Avegawanderer, 2022-12-03:
Hello, I've run into the same problem with g-code gerenrated with FlatCAM. After removing G01 error is still there. 
Can anyone help, please?
[main.GBL_iso_combined_cnc.txt](https://github.com/winder/Universal-G-Code-Sender/files/10147169/main.GBL_iso_combined_cnc.txt)

ThiagoTuareg, 2023-05-01:
Olá, estou com o mesmo erro. Gero o arquivo Gcode apartir do flatcam com o processador (grbl no M6).
Abro o arquivo gcode e altero todas as linhas G01 Fxx para a coordenada abaixo G01 Xxx Yxx Fxx. Porem o erro persiste em nao aplicar no auto nivelamento.
Universal Gcode sender platform

ThiagoTuareg, 2023-05-01:
Olá pessoal eu consegui utilizar o auto leve excluindo todas as linhas de código g01 fxx, so assim eu comsegui aplicar

breiler, 2024-02-07:
I can't reproduce the problem in UGS 2.1.4