winder/Universal-G-Code-Sender #1432: Error Sending files to arduino with Universal G-code Sender

AndrwPerez, 2020-07-29:
## Feature request
Error Sending files to arduino with Universal G-code Sender 

## Problem description 
When i try to send the files to my arduino only send a few and stops, if i click "Send" again pops de error "Error while starting file stream"

### Expected Behavior
Sending all the files 

### Actual Behavior
Sending like 22 or 26 files of 700 and stops

### Steps to Reproduce the Problem
 1. Open the program and the arduino with the GRBL Library 
 2. Try to send a .cnc file to arduino
  1.
  1.

### Specifications
#### Version
UGS Platform 2.0 
--

#### Operating system
Windows 10
 

#### Platform

#### Other

AndrwPerez, 2020-07-29:
[image]

breiler, 2020-07-29:
My best guess is that you are getting interference from stepper motors, spindle or dust collector which causes your Arduino to reset and UGS looses connection to it. Try running your gcode program without any peripherals connected or powered to rule out if this is the case.

AndrwPerez, 2020-07-29:
> My best guess is that you are getting interference from stepper motors, spindle or dust collector which causes your Arduino to reset and UGS looses connection to it. Try running your gcode program without any peripherals to the Arduino connected or powered to rule out if this is the case.

My arduino is only connected to the PC, so i don't think so

breiler, 2020-07-29:
You only have the Arduino connected and nothing else?

What kind of Arduino Uno are you using, is it genuine one or a cheap China variant? 
Are you using a proper shielded USB cable?
Which version of GRBL are you using?

AndrwPerez, 2020-07-29:
> You only have the Arduino connected and nothing else?
> 
> What kind of Arduino Uno are you using, is it genuine one or a cheap China variant?
> Are you using a proper shielded USB cable?
> Which version of GRBL are you using?

Yes, when i upload the files i have the Arduino only connected to the PC
I'm using an original Arduino Uno
And i'm using this GRBL
[GRBL-Arduino-Library-master.zip](https://github.com/winder/Universal-G-Code-Sender/files/4996560/GRBL-Arduino-Library-master.zip)

breiler, 2020-07-29:
That GRBL version looks like an old "0.8c". I think for a better experience you should try the latest:
https://github.com/gnea/grbl/releases/tag/v1.1h.20190825

Is there any reason you are using that version?

AndrwPerez, 2020-08-02:
There isn't nay reason why i'm using the 0.8c, but i downloaded the new version and now the sending of files doesn't even start for some reason

AndrwPerez, 2020-08-02:
[image]

breiler, 2020-08-02:
You are connecting with baud rate 9600, it will work better with 115200
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands#getting-started

closed 2020-08-07 by breiler