winder/Universal-G-Code-Sender #1433: Raspberry Pi Install Instructions need updating

MonkeyManGuy, 2020-08-05:
## Problem description 
While following the Rasp Pi instructions step 3 includes options that do not appear on Raspberry Pi. We are looking for "7 Advanced Options"->"A7 GL Driver"->"G3 GL" (in my case)

The reality of the current OS for Pi is that A8 Is for the GL Driver, however when on this page "G3 GL" is not an option. Please see the picture attached:

### Expected Behavior
Progress through step 3 of install instructions and locate all categories and options described as nessacary in the instructions.

### Actual Behavior
Step 3 contains a category level deveation from the current operating system release, on the subsequent page the needed option is not listed.

### Steps to Reproduce the Problem

MonkeyManGuy, 2020-08-05:
[image]
[image]

zoominbc, 2020-08-24:
I believe that the G3 option is supported for a Pi 3 but not for a Pi 4. The G2 option Fake KMS does work.

MonkeyManGuy, 2020-12-13:
> I believe that the G3 option is supported for a Pi 3 but not for a Pi 4. The G2 option Fake KMS does work.

So this means Pi 4 users must choose the G1 option?

sepiroth887, 2022-11-15:
use nightly as of [d432d5e](https://github.com/winder/Universal-G-Code-Sender/commit/d432d5ec3891f8347a457e1ce13c0b4a244a9c37)  which should not require any changes in Pi 4 gpu drivers. (tried with software and FakeKMS)

breiler, 2023-01-24:
I have updated the install instructions. Though only for 32-bit systems as that was the only one I could get to work reliably.

closed 2023-01-24 by breiler