winder/Universal-G-Code-Sender #1434: Unable to Resize Windows

tfjield, 2020-08-07:
## Problem description 
Depending on the window layouts, sometimes I'm unable to resize the windows.  Sometimes they have minimum sizes that are too large, and sometimes they're not resizeable at all, depending on what windows are around them.  (Probably because several windows have minimum sizes too large to fit in the column.)

### Expected Behavior
I should be able to shrink the window down until the blank space around the text is almost invisible.  (As a side note, it would also be nice to see what's at the bottom of the DRO window...  Any way to get rid of the help button in Common Actions to free up that real estate?)

### Actual Behavior
In the following layout, I am completely unable to change the height of the State window, the Overrides window, or the Send Progress Window; their relative sizes are fixed.  I can adjust the widths, though.
[image]

In the following layout I am unable to shrink the height of the State window or the Send Progress window, and thus I can't shrink the Console window, either.  I can adjust widths, though.
[image]

### Steps to Reproduce the Problem
Place the window as shown in the images and try to resize them.

### Specifications
#### Version
UGS Platform 2.0 - Nightly Build Jun 25, 2020 (Also tried July 29 nightly)
Look and Feel is set to Windows.

#### Operating system
Windows 10

JacksCrafts, 2020-08-07:
I agree they could use some work.
I notice this with the probe module, it seems to have a very wide min.
On my layout I have the Jogging on the right, and that allows my DRO to be fully seen.
[image]

tfjield, 2020-08-07:
I don't know how popular this would be, but I suggest the easiest way to solve this problem is to remove the minimum window sizes completely.  (Or make them ridiculously small; just large enough that the window doesn't get lost.)  Once the controls have shrunk down as much as they can to accommodate the window size, then they would just get clipped as is typical for a lot of Windows applications.

That leaves it up to the user to decide whether he or she is happy with the clipping.  I would be happy to clip off the bottom of the Common Actions window, for example, since I don't need the help button.  Others might like it there.  (Lots of CNC'ing and beer, hard to remember what the words mean...  lol)  At least the user could decide.

I'm still playing with layouts, but I really like the Jog Controller close to the DRO.  But if I just have two of them on the left, it wastes a lot of valuable window space.

whitetd, 2020-08-08:
Welcome to java.