winder/Universal-G-Code-Sender #1436: User interface broken

itsharrie, 2020-08-17:
## Feature request

## Problem description 
The user interface does not initiate properly when starting up ugsplatform. 

[image]

### Expected Behavior
Properly show the window.

### Actual Behavior
UI elements are seemingly randomly refreshed when clicking anywhere on the window.

### Steps to Reproduce the Problem
Open ugsplatform 
  1.
  1.
  1.

### Specifications
#### Version
Stable version 2019-08-14 or nightly version 2020-08-10 (same problem)

#### Operating system
Linux Mint 20

#### Platform
n/a

#### Other

breiler, 2020-08-18:
Did you try the version with bundled java from here:
https://github.com/winder/Universal-G-Code-Sender#downloads

breiler, 2020-09-11:
Closing as there was no response from user.

closed 2020-09-11 by breiler