winder/Universal-G-Code-Sender #1439: G-Code Feedrate Command is ignored

Felix-Hilken, 2020-09-05:
Dear Community, 

I'm using this program to run a 2 Axis CNC Hotwire cutter. For hotwire cutting a constant feed rate is crucial. I've programmed my G-Codes to use a F200 Command to run at 200 mm/min, as it has been determined optimal before I switched to a new GRBL controller. 

When using the Jog Mode and putting in a Feedrate of 200, the machine accepts the input and runs at a constant 200 mm/min. It slows down both axis on diagonal jogs (as it should) to reach a rate of 200 mm/min for all movements. 

When I load my G-Code files and run them however, the machine sprints through the entire code at the maximum allowed speed for each axis from the configuration file. My maximum axis speed is 500 mm/min for X and Y, resulting in diagonal speed of up to 700 mm/min. At these speeds the wire just gives in and snaps. 

I assume this problem originates somewhere within the program part responsible for converting the G-Code information into commands that get sent into the buffer. Neither adding an F200 command to every line of the G-Code, nor inputting a Feedrate using the "state" window changes anything. 

It might be my G-Code that is faulty, but I doubt that it is as it's geometry is visualised fine. I'll just past one of them in here. Maybe I'm just stupid and messed something up in there.

I hope you can help me with this problem or work out a fix. 

Cheers, 
Felix

Sample G-Code:
N0010 G21G90X320.Y50.3535 F200
N0020 X228.F200
N0030 X183.4765 Y50.9579 F200
N0040 X148.1059 Y51.3228 F200
N0050 X134.4323 Y75. F200
N0060 X38.43 F200
N0070 X25.1182 Y51.9494 F200
N0080 X0.0 Y51.9674 F200
N0090 X-25.1182 Y51.9494 F200
N0100 X-38.43 Y75. F200
N0110 X-134.4323 F200
N0120 X-148.1059 Y51.3228 F200
N0130 X-194.9246 Y50.819 F200
N0140 X-228. Y50.3535 F200
N0150 X-320. F200
 
N0160 M02

breiler, 2020-09-06:
Which version of UGS and GRBL are you using?

svenhb, 2020-09-06:
G0 (rapid mode) is the default motion mode after reset. I don't see a G1 command in your code...
Check
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Commands#g---view-gcode-parser-state
http://linuxcnc.org/docs/html/gcode.html

Felix-Hilken, 2020-09-07:
Oh my god I'm an idiot. Talk about not seeing the forrest for the trees. Yeah I just added a G1 line and it works just fine. Thanks for the help.

closed 2020-09-07 by Felix-Hilken