winder/Universal-G-Code-Sender #1441: UGS Platform not responding after launch

StevenDedeurwaerder, 2020-09-17:
Hello,

I'm completely new to UGS Platform, this is my first time using it. UGS Classic seems to work pefectly fine for me. However i can't work with UGS Platform.

## Problem description 
I've come across an issue similar to #1337 , after i launch UGS Platform via  ugsplatform.exe or ugsplatform64.exe i get stuck in a screen as shown in #1337 , where no buttons are responsive. I have tried this with the nightly build, the stable build and the build with java preconfigured with the exact same results. I am fairly certain i updated java to the right version.
So far i have tried:
-deleting .ugs and .ugsplatform folders
-launching ugsplatform in every compatibility mode (windows xp, windows vista,..)
-turning off my antivirus (Bullguard)

### Expected Behavior
Launching UGS Platform.

### Actual Behavior
Launching an unresponsive window that has to be closed with task manager.

#### Version
UGS Platform Classic/Nightly/Preconfig Java

#### Operating system
Windows 10

#### Other
These are the log files.
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/5235522/messages.log)

breiler, 2020-09-17:
Thanks for an excellent issue report! Unfortunately the log did not give any clues on what is happening. As similar issues has revolved around the AV-software I still suspect that it is still the culprit. Does Bullguard have any logs of its activitiy?

ShortCircc, 2020-09-18:
I think this should be the part of the log file where i open ugsplatform and close it off with taskmanager.

2020/09/18 17:23:23.548 | [PROC] WhiteListed    [PID =  3364 ] [Path = C:\Users\Gebruiker\Desktop\ugs-platform-app-2.0-SNAPSHOT\ugsplatform\bin\ugsplatform.exe ]  [PUNQ = 5695126637860 ] [PPID =  4676 ]
2020/09/18 17:23:23.556 | [PROC] ProcessExit    [PID =  3364 ] [Path = C:\Users\Gebruiker\Desktop\ugs-platform-app-2.0-SNAPSHOT\ugsplatform\bin\ugsplatform.exe ]  [PUNQ = 5695126637860 ] [PPID =  4676 ]
2020/09/18 17:23:23.564 | [PROC] ProcessCreated [PID = 10212 ] [Path = C:\Windows\System32\consent.exe ]  [PUNQ = 5699421612004 ] [PPID = 11596 ]
2020/09/18 17:23:25.822 | [PROC] ProcessExit    [PID = 10212 ] [Path = C:\Windows\System32\consent.exe ]  [PUNQ = 5699421612004 ] [PPID = 11596 ]
2020/09/18 17:23:25.844 | [PROC] WhiteListed    [PID = 14656 ] [Path = C:\Users\Gebruiker\Desktop\ugs-platform-app-2.0-SNAPSHOT\ugsplatform\bin\ugsplatform.exe ]  [PUNQ = 5703716583744 ] [PPID =  4676 ]
2020/09/18 17:23:27.451 | [PROC] ProcessCreated [PID =  9576 ] [Path = C:\Windows\System32\SearchProtocolHost.exe ]  [PUNQ = 5708011545960 ] [PPID =  6348 ]
2020/09/18 17:23:49.291 | [PROC] ProcessCreated [PID =  7988 ] [Path = C:\Users\Gebruiker\AppData\Local\Temp\jogamp_exe_tst8076457640711040836.exe ]  [PUNQ = 5712306511668 ] [PPID = 14656 ]
2020/09/18 17:23:49.292 | [PROC] New HardLink   [PID =  7988 ] [Path = C:\ProgramData\BullGuard\SentryTemp\jogamp_exe_tst8076457640711040836.exe.F24636085173170FB825B919007D6E5D ]  [PUNQ = 5712306511668 ] [PPID = 14656 ]
2020/09/18 17:23:49.296 | [UNTRUSTED] FileInfo  [PID =  7988 ] [Path = C:\Users\Gebruiker\AppData\Local\Temp\jogamp_exe_tst8076457640711040836.exe ] [PPID = 14656 ] [PUNQ = 5712306511668 ]
                          [i] [SHA256 = 69174E37703A7027509DEA3942662A846FB733A3D22D5CEA63679954BE3C183E ] [MD5 = 676A4D40F0DB77A49D4C875A85584CB5 ]
2020/09/18 17:23:49.293 | [i] DigitalSignature UNSIGNED in 0(ms)
2020/09/18 17:23:49.293 | [i] KnapResult NOT_FOUND in 0(ms)
2020/09/18 17:23:49.293 | [i] CalculateFlags SUCCESS in 0(ms)
2020/09/18 17:23:49.297 | [e] SentryEyeClient::CommandEvaluate failed to write command ErrorCode=9
2020/09/18 17:23:51.157 | [PROC] ProcessCreated [PID =  1560 ] [Path = C:\Windows\System32\wbem\WmiPrvSE.exe ]  [PUNQ = 5716601472536 ] [PPID =    84 ]
2020/09/18 17:24:01.672 | [PROC] ProcessExit    [PID =  7988 ] [Path = C:\Users\Gebruiker\AppData\Local\Temp\jogamp_exe_tst8076457640711040836.exe ]  [PUNQ = 5712306511668 ] [PPID = 14656 ]
2020/09/18 17:24:01.699 | [PROC] ProcessCreated [PID = 10824 ] [Path = C:\Users\Gebruiker\AppData\Local\Temp\jogamp_0000\jogamp_exe_tst1646051621730856508.exe ]  [PUNQ = 5720896449096 ] [PPID = 14656 ]
2020/09/18 17:24:01.700 | [PROC] New HardLink   [PID = 10824 ] [Path = C:\ProgramData\BullGuard\SentryTemp\jogamp_exe_tst1646051621730856508.exe.4A427CAE06C945DE764D416B2B03DDAB ]  [PUNQ = 5720896449096 ] [PPID = 14656 ]
2020/09/18 17:24:01.703 | [UNTRUSTED] FileInfo  [PID = 10824 ] [Path = C:\Users\Gebruiker\AppData\Local\Temp\jogamp_0000\jogamp_exe_tst1646051621730856508.exe ] [PPID = 14656 ] [PUNQ = 5720896449096 ]
                          [i] [SHA256 = 69174E37703A7027509DEA3942662A846FB733A3D22D5CEA63679954BE3C183E ] [MD5 = 676A4D40F0DB77A49D4C875A85584CB5 ]
2020/09/18 17:24:01.701 | [i] DigitalSignature UNSIGNED in 0(ms)
2020/09/18 17:24:01.701 | [i] KnapResult NOT_FOUND in 0(ms)
2020/09/18 17:24:01.701 | [i] CalculateFlags SUCCESS in 0(ms)
2020/09/18 17:24:01.704 | [e] SentryEyeClient::CommandEvaluate failed to write command ErrorCode=9
2020/09/18 17:24:01.988 | [PROC] ProcessCreated [PID = 16288 ] [Path = C:\Windows\System32\svchost.exe ]  [PUNQ = 5725191421856 ] [PPID =   868 ]
2020/09/18 17:24:02.020 | [PROC] ProcessCreated [PID =  4480 ] [Path = C:\Windows\SysWOW64\WerFault.exe ]  [PUNQ = 5729486377344 ] [PPID = 16288 ]
2020/09/18 17:24:03.821 | [PROC] ProcessCreated [PID =  4828 ] [Path = C:\Windows\System32\wermgr.exe ]  [PUNQ = 5733781344988 ] [PPID =  1632 ]
2020/09/18 17:24:03.830 | [PROC] ProcessExit    [PID =  4480 ] [Path = C:\Windows\SysWOW64\WerFault.exe ]  [PUNQ = 5729486377344 ] [PPID = 16288 ]
2020/09/18 17:24:03.854 | [PROC] ProcessExit    [PID = 14656 ] [Path = C:\Users\Gebruiker\Desktop\ugs-platform-app-2.0-SNAPSHOT\ugsplatform\bin\ugsplatform.exe ]  [PUNQ = 5703716583744 ] [PPID =  4676 ]
2020/09/18 17:24:05.865 | [PROC] ProcessExit    [PID =  4828 ] [Path = C:\Windows\System32\wermgr.exe ]  [PUNQ = 5733781344988 ] [PPID =  1632 ]
2020/09/18 17:24:09.748 | [PROC] ProcessCreated [PID =  7084 ] [Path = C:\Windows\System32\notepad.exe ]  [PUNQ = 5738076314540 ] [PPID =  4676 ]

breiler, 2020-09-21:
Was this created after you turned off the AV-software?

It looks like it's blocking the visualizer library  (or at least flagging as untrusted) we are using (jogamp).
```
2020/09/18 17:23:49.292 | [PROC] New HardLink [PID = 7988 ] [Path = C:\ProgramData\BullGuard\SentryTemp\jogamp_exe_tst8076457640711040836.exe.F24636085173170FB825B919007D6E5D ] [PUNQ = 5712306511668 ] [PPID = 14656 ]
2020/09/18 17:23:49.296 | [UNTRUSTED] FileInfo [PID = 7988 ] [Path = C:\Users\Gebruiker\AppData\Local\Temp\jogamp_exe_tst8076457640711040836.exe ] [PPID = 14656 ] [PUNQ = 5712306511668 ]
```

I have no idea how to proceed with this, not much we can do from UGS.

ShortCircc, 2020-09-21:
Those were the logs with the AV-software on, i don't have any logs when it's turned off. Adding relevant folders to exceptions of the AV doesn't have much effect either.
Jogamp also keeps running after force closing ugsplatform. 
[image]
These are multiple attempts with different versions.

breiler, 2022-12-16:
You could try the latest nightly build (or version 2.0.13 when it gets released). We have updated the graphics library which might fix the problem. Closing as we can't reproduce the error.

closed 2022-12-16 by breiler