winder/Universal-G-Code-Sender #1443: screen resolution issue on raspi display 800x480

chpatil, 2020-09-28:
## Feature request

## Problem description 

The problem is the that i am unable to scale the ugs the screen it goes out of the bounds all other programs are scaled in that display but ugs is unable to scale in that display

### Expected Behavior

it is expected to scale the ugs

### Actual Behavior

the ugs is not gerting scaled
[image]

#### Version
UGS Platform 2.0 - Nightly Build Aug 14, 2019

#### Operating system

Linux on RaspberryPI

#### Platform
GRBL 1.1f

#### Other

breiler, 2021-01-08:
We will not fix this in the old classic edition. 
You should try the platform edition instead. Even if it's not perfect the windows can be rearranged to fit a screen with lower resolutions.

breiler, 2021-01-08:
Forgot to mention that we recently fixed a problem with the JDK which caused the program to use 100% of the CPU. Make sure you use the latest nightly build. And if you find any problems with the UI in lower resolutions you are welcome to file a new issue.

closed 2021-01-08 by breiler