winder/Universal-G-Code-Sender #1447: Serial Connection for DIY Pendant

wile1411, 2020-10-12:
## Feature request
Just an idea for a plugin that suggests a way for people to extend the capabilities of their machine if desired. 

The suggestion is to have a plugin that lets you choose a COM port for a micro processor to interface with certain API functions. My suggested initial use case would be to use the proposed serial connection to avoid all the issues that come with a keyboard emulator pendant solution. (Mouse focus, action latency, shortcut planning, etc...)

Instead, the connection would be use a small Arduino to issue commands directly via the API. The code from the plugin side would need to be able to pass sent Gcode commands to the machine and return replies with a return message. 
response = UGSserial.writeConsoleCommand("$H")
MachineStateParser(UGSserial.writeConsoleCommand("$G"))
PositionParser(UGSserial.writeConsoleCommand("?"))

This way people could make & code their own version of a pendant and add any button, jog wheel, status change they wanted. Granted it's similar to what's available with the built-in UGS pendant, but this would allow people to come up with their own (ideally shareable) solutions and hopefully take some of that development load off you to come up with a one-size fits all solution.

breiler, 2020-10-12:
I'm not exactly sure I understand 🤔 

Is the feature you are looking for a function that will make the asynchronized `command + reply` to become synchronized?

For instance, today if you want to send a command and find out the results of that command you would need to implement and register an [ControllerListener](https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-core/src/com/willwinder/universalgcodesender/listeners/ControllerListener.java#L31). 

But with the suggested **UGSserial.writeConsoleCommand("command")** it would block the operation and await and return the response from the controller?

wile1411, 2020-10-12:
Ah - I see what I did and that was mainly my poor attempt at writing the suggestion while having my focus too much on the possibilities.
So, no the operation on the microprocessor would definitely NOT be blocked and the process would have to have a listener function. Also you are correct in that the function UGSserial.writeConsoleCommand("command") is completely wrong for serial messaging. (I get stuck thinking procedurally at times)
It'd probably be more like: (only trying to emphasise the none blocking) 
loop(){
  if(cmdtriggered) sendUGScmd(cmd)
  if(responseavailable) readUGSresponse()
}
sendUGScmd(cmd){
  Serial.write(packet, headers, len, etc...) 
  resetsendtriggers()
}

The function just boils down to being able to program a custom microcontrollor pendant just using the command line interface of UGS. This idea is born from the UNO pins are maxed out, my desire to not change my Uno to a Mega AND I also still want to keep and make use of the USG UI as it has everything available on display.

From my reading, people have managed to get DIY pendants to work using keyboard emulation, but there also seems to be a catch. I figure I can get more out of pendant given a microprocessor that can interface with the command line directly without relying on application focus, as well as being able to parse the responses for information / machine status. (I'm aware I'd have to write a ton of response parsing logic. But I only have a finite set of buttons / switches I'd want to add, so I know it's a doable task.
Eg Switch MPG scale (0.1,1,10,100), Switch MPG selection (X,Y,Z,Feed,Speed,Jog), Switch for Offsets(M,1,2,3,4,5) Plenty of buttons (Home Machine, Zero All, Zero X/Y/Z/F/S, Goto XYZ Zero, Goto XYZero, Dust, Light, Feed Hold, Start/Resume, Reset/Abort, ProbeZ, ProbeAll, Unlock/SoftReset, a few macro buttons). All the usual stuff.

breiler, 2020-10-12:
Ok, so you would like to integrate with UGS using serial communication, neat idea. I don't have the time to work on a feature like this right now (maybe someone else is interested?), so in the meantime I can give you some other options.

**Option 1** - Using a "Man in the middle" hardware which will proxy all communications with the controller (i.e. GRBL) and also send its own commands. Even if it should be possible, it is not a good idea. Because UGS is trying to keep track of all the commands that are sent and the responses received. If it gets a bunch of "OK" responses it doesn't know anything about things will become messy internally in UGS and errors may occur.

**Option 2** - Use the existing REST/RPC API. The documentation for it is a bit lacking but should be pretty easy to understand if you [read the code](https://github.com/winder/Universal-G-Code-Sender/tree/master/ugs-pendant/src/main/java/com/willwinder/universalgcodesender/pendantui/v1/resources). If you activate the "Pendant" in UGS it will start listening to web calls.
The only downside with this solution is that it requires polling for states and the protocol will become very chatty. A solution to this would be WebSockets, but we are not there yet.

**Option 3** - Use a [gamepad or joystick](https://github.com/winder/Universal-G-Code-Sender/wiki/Usage#gamepad-and-joystick). This only allow for one way communication, but could maybe be combined with option 2. The big advantage with this option is that it allows for using analog controls.

wile1411, 2020-10-12:
Thanks for the ideas.
For option 1, I'm already investigating something like the XLcd (https://wiki.shapeoko.com/index.php/XLCD) as a hardware proxy as you suggested. But as you also mentioned, I'd rather not be generating unknown responses back to UGS. I did see someone project that added a 1-way connection with a diode on the line by tapping into the RX/TX lines. But, again I'd prefer not to generate unknown responses. Anything that's effectively the same as a user manually keying the commands into the UGS console is the goal. That way, UGS knows about the request being sent.

For option 2, Is that something like how this project was implemented? (https://discuss.inventables.com/t/prototype-ugs-pendant/47171/3) Thanks for the link for the REST/RPC. I'll read those today.

For option 3, I'd probably fall back to a keyboard emulator that a gamepad. Personal preference more than anything, but I know other people prefer that control type.

winder, 2020-10-12:
You can also tap into the communication later of UGS to implement special
logic before paying it to the GRBL controller. There wasn't much interest
at the time but we added the hook for XLCD a while back if you want to
experiment with it:

https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-core/src/com/willwinder/universalgcodesender/XLCDCommunicator.java

winder, 2020-10-12:
"option 2" is what the project you linked uses, and the REST/RPC services
in UGS are a lot better today than they were back then. That approach is
going to be a lot simpler than the man-in-the-middle pendant.

closed 2020-11-16 by breiler