winder/Universal-G-Code-Sender #1456: Program fails when the gui with visualzer attempts to load

DUVIVI, 2020-10-14:
## Feature request

## Problem description 

I am attempting to run UGSPlatform on a 2015 macbook Pro  runing OS 10.14.6.  After installing Java 1.8.0+ and then installing UGSPlatform I launch the unix verison of UGSplatform from the Bin folder.  The initial text window is opened and the the window with the UGS Graphic appears and the Graphic image is opened in that window with a progress bar.  Then the gui for UGS opens but only has a small window within the Primary window with the Java Logo and the statement...

This version of Universal Gcode Sender is a nightly build.
It contains all of the latest features and improvements,
but may also have bugs that still need to be fixed.

If you encounter any problems, please report them on github.

At this point the gui is erased and a "Problem report" from apple is generated.  The statement on the report is that **ugsplatform quit unexpectedly.** I have completely removed Java and UGS platform files including all files in the Library path several times and reinstalled them and the behavior is the same.  I have also installed the files on a mac runing the same operating system that has nothing else on it so it is completely clean with the same result.  What really makes this confusing is that this exact set up ran without problems 6 months ago and I was merely updating the software.  I have tried to include a video from my phone of the progress of this process because it completed so quickly and didn't remain long enough to capture as an image, unfortunately you don't support any video formats.  I should mention that I installed UGS Classic and it runs ok until I select the button for the visulizer and the program terminates in exactly the same way.  Last clarification this occurs whether or not the computer is connected to the Arduino.

### Expected Behavior

### Actual Behavior

### Steps to Reproduce the Problem

  1.launch UGS platform
  1.
  1.

### Specifications
#### Version

UGS Platform 2.0 - Nightly Build Oct 13, 2020

#### Operating system

Mac OS Mojave 10.14.6

#### Platform

Arduino 2560

#### Other

This is an image of the Problem report from apple taken on the screen of the Mac 
[image]

breiler, 2020-10-14:
It looks like its using the wrong version of Java (accessing it from `/Library/Internet Plug-Ins/JavaAppletPlugin.plugin/Contents/Home/bin/java`). 

Are you using the Mac-version with a bundled Java, otherwise download it from here:
https://github.com/winder/Universal-G-Code-Sender#downloads

DUVIVI, 2020-10-15:
I wanted to report that I downloaded the combined UGS Platform  with the recommended Java as you suggested.  It works perfectly!  Thank you for entertaining my posting I really appreciate it.  I attempted much to resolve it myself.  I have to admit I was convinced that it was a problem with Java which appears to be true.  Thank you for your email so much.  I will definitely be contributing to the site again.  If there is anything I can contribute please let me know.

https://github.com/winder/Universal-G-Code-Sender#downloads

closed 2020-10-16 by breiler