winder/Universal-G-Code-Sender #1463: Open Recent Files option (Feature Request)

khturnings, 2020-11-01:
Feature request
Can you add a item to the file drop down menu to show a list of recently opened files.

Problem description 
When working on a batch of jobs over a few days each time I need to open a file I need to scroll through all my Gcode files in the directory they are stored in.

Expected Behaviour
Use File -> Open Recent -> List last xx number of jobs opened.

Actual Behaviour
Currently not available

winder, 2020-11-01:
The start page has a recently opened files tab.

khturnings, 2020-11-01:
So it does - I have never really used the tabs on the start page.
Will close this as you have plenty of other more important issues to resolve.

closed 2020-11-01 by khturnings