winder/Universal-G-Code-Sender #1471: Zero/Insert doesn't reset all zero

Smoghog, 2020-11-11:
## Problem description 
I've read in several places that zero/insert is supposed to reset all to zero when using keyboard commands, but I've found that it isn't working on mine. All other jog commands work as expected.

### Expected Behavior
Nothing happens when pressing zero/insert, whether numlock is on or off.

Am I alone in this issue? Or did this shortcut get removed somewhere along the way?

breiler, 2020-11-11:
Hi @Smoghog, what I can see, that combination is disabled (commented out) for some reason:
https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-classic/src/main/java/com/willwinder/universalgcodesender/MainWindow.java#L1564

Is there any reason you aren't running the platform edition, there is a lot more flexibility on keyboard mappings there.

Smoghog, 2020-11-11:
> Hi @Smoghog, what I can see, that combination is disabled (commented out) for some reason: [...]

I'm using Platform myself, I absolutely love it. But I am developing a pendant using basic keyboard commands and I've been asked be a couple people if I will make it Classic compatible in addition to Platform, so I'm looking into it currently. I'm not versed in Java development, is there any way I can re-enable that feature since it is in there but simply disabled? Perhaps I could then create my pendant and add the caveat that this needs to be done in order for that function to work.

PR #1472 (2020-11-15): Reactivated keyboard shortcut for resetting position
Fixes #1471
closed 2020-11-15 by breiler