winder/Universal-G-Code-Sender #1474: Graphics and code still running after job completed on laptop. Graphics to slow on laptop UGS 2.0 nightly  Bob's CNC Evolution 4

Tenpin, 2020-11-16:
**Describe the bug**
A clear and concise description of what the bug is.

**To reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version**
[e.g. UGS Platform 2.0.6, UGS Classic 1.0.9]

**Hardware**
[e.g. GRBL 1.1f, TinyG,  SmoothieBoard]

**Operating system (please complete the following information):**
 [e.g. Windows 10, MacOSX, Ubuntu 20.04 ]

**Additional context**
Add any other context about the problem here.

breiler, 2020-11-16:
What is the build date of your nightly build (is displayed in the title)?

Tenpin, 2020-11-16:
August 14th 2019

breiler, 2020-11-16:
That is a known old issue. Please upgrade to the newest release from here:
https://github.com/winder/Universal-G-Code-Sender#downloads

Tenpin, 2020-11-16:
Okay thanks for the quick response