winder/Universal-G-Code-Sender #1475: Unable to locate a Java Runtime that supports (null).

Doc1455, 2020-11-17:
**Describe the bug**
Last login: Tue Nov 17 09:53:13 on console
/Users/admin/Downloads/ugsplatform/bin/ugsplatform ; exit;
admin@sever ~ % /Users/admin/Downloads/ugsplatform/bin/ugsplatform ; exit;
The operation couldn’t be completed. Unable to locate a Java Runtime that supports (null).
Please visit http://www.java.com for information on installing Java.

**To reproduce**
When i try to run the ugsplatform it opens a UGS window and then closes
And this is in the terminal window

**Expected behavior**
Program to run

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version**
 UGS Platform 2.0.6

**Hardware**
GRBL 1.1f

**Operating system **
  MacOSX, 

**Additional context**
Last login: Tue Nov 17 09:53:13 on console
/Users/admin/Downloads/ugsplatform/bin/ugsplatform ; exit;
admin@sever ~ % /Users/admin/Downloads/ugsplatform/bin/ugsplatform ; exit;
The operation couldn’t be completed. Unable to locate a Java Runtime that supports (null).
Please visit http://www.java.com for information on installing Java.
Ive installed Java * several times and still get this error

breiler, 2020-11-17:
Did you download the MacOSX package?
https://github.com/winder/Universal-G-Code-Sender#downloads

Doc1455, 2020-11-17:
Yes i did now, and it works now. 
Thank you!

closed 2020-11-20 by breiler