winder/Universal-G-Code-Sender #1477: UGS Platform on RasPi large GCode file problem

phil-barrett, 2020-11-19:
**Describe the bug**
Loading a very large GCode file chokes Platform running on RasPi.

**To reproduce**
Steps to reproduce the behavior:
1. Start UGS Platform (I used build DL'd on 11/18/2020), connected to grbl board (I am using grblHAL but don't think that matters in this case)
2. Load "concentric circles as arcs fr5000.cnc" (see attached zip file)
3. Note that the machine has become extremely unresponsive. Mouse cursor movement is delayed by 5-10 second.
4. Let it run 8 hours and note it is still unresponsive.
5. Power cycle your Pi

**Expected behavior**
The GCode will load.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version**
From the about box.
```
Product Version: Universal Gcode Platform 20201002
Java: 1.8.0_212; OpenJDK Client VM 25.212-b01
Runtime: OpenJDK Runtime Environment 1.8.0_212-8u212-b01-1+rpi1-b01
System: Linux version 5.4.72-v7+ running on arm; UTF-8; en_US (ugsplatform)
User directory: /home/pi/.ugsplatform/2.0.6/dev
Cache directory: /home/pi/.ugsplatform/2.0.6/dev/var/cache      

```
**Hardware**
RasPi 3 model B running raspi buster (2020-02-12-rpd-x86-buster.iso image), 
grblHAL breakout board running grblHAL with COMPATIBILITY_MODE 1 set.

**Additional context**
Running same file on ugs platform on a Win10 PC has no problems.

Here is the file
[concentric circles as arcs fr5000.zip](https://github.com/winder/Universal-G-Code-Sender/files/5569444/concentric.circles.as.arcs.fr5000.zip)

breiler, 2020-11-21:
Wow that was an evil test on the little pi. 😄 

The arcs will be expanded to lines using with a segment length of 0.3mm which would generate about 7.3 million line segments. That is a lot... 

We could probably try to parse the extent of the work before loading it as a model to get some estimate on a suitable arc length.

phil-barrett, 2020-11-21:
Yeah, diabolical. It tends to choke renderers as well.  Definitely not a realistic test, though. UGS seems to handle my other tests. If you guys want to parse far enough to say "won't run" or even just close without fixing, I would not think less of you.