winder/Universal-G-Code-Sender #1479: Small size drawed

ladyfirst2010, 2020-11-23:
Hi,

I export .gcode from inkscape then run with UGS but it's very small image drawed.

Any suggestion for me to fix it please ?

ladyfirst2010, 2020-11-23:
Example gcode :

M3
G90
G21
G1 F1000
G1  X2.1335 Y29.4418
M5 S90
G4 P0.5
G1 F500.000000
G1  X56.7502 Y29.4418
G1  X56.7502 Y2.7735
G1  X2.1335 Y2.7735
G1  X2.1335 Y29.4418
M3
G4 P0.5
G1 F1000
G1 X0 Y0

breiler, 2020-11-24:
I'm not sure what you are expecting, but the gcode is a rectangle that is approximately 56x30mm. 
What was the expected size?

ladyfirst2010, 2020-11-25:
Hi.

Thanks for your reply. I expect it much bigger because I draw it in A4
size.
Dont know what cause the problem, Inkscape or the MI extension.

breiler, 2020-11-25:
My guess is it's in the MI extension, you will need to check its manual.
I'm closing this issue as it is a bit beyond the scope of UGS. 

Good luck!

closed 2020-11-25 by breiler

ladyfirst2010, 2020-11-25:
Thank a lot for your help