winder/Universal-G-Code-Sender #1480: UGS platform 2.0.61 [Error] Error while processing response

wolf77957, 2020-11-24:
hello and in the meantime congratulations, I have a CNC 2418 Woodpecker 0.8 with GRBL 0.9j, so far I have used GRBL Controller 0.8 or candle 1.0 and all is well. Now I have installed U-gcode sender 2.06.61 attracted by the greater possibilities (auto leveling etc). I can move the cutter in X Y and Z. unfortunately after the connection it signals me during the movement it signals  in n CONSOL E**"[Error] Error while processing response"** continuously. any enlightening ideas? I am attaching diagnostic error .thanks in advance
[image]
[image]
[image]

**Version**
UGS Platform 2.0.61

**Hardware**
GRBL 109g

**Operating system **
 Windows 10

wolf77957, 2020-11-27:
solved, an update to gbrl 1.1h was enough

breiler, 2020-11-27:
@wolf77957, great!
I would like to keep this issue open as GRBL 0.9 should still be supported.

geko7844, 2021-04-01:
> Same problem with grbl 1.1H
> [Error] Error while processing response <<Idle|Pos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Ov:100,100,100>>
> Can't find the answer
> Thanks

I  had the same issue with UGS. I have uninstalled and re-installed JavaScript from my PC, as well as disable a lot of unnecessary programmes at start up and it seems to have made a big difference to the number of errors.  Might be completely unrelated, but worth a try.

c0usin, 2022-01-20:
Helo, I have the same issue with my new MKS DLC v2.1 board and UGS v2.0.9.
Can't figure out what's happening, used with Arduino Uno and CNC shield without this issue, now after I connect to the board with COM getting this error until I disconnect.

I tried to report the problem from program, don't know if it was sent or no, but here is my log:

INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Connecting to GRBL on port COM3
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Applying settings.
INFO [com.willwinder.universalgcodesender.AbstractCommunicator]: Connecting to controller using class: JSerialCommConnection with url jserialcomm://COM3:115200
SEVERE [global]
java.lang.NullPointerException
	at com.willwinder.universalgcodesender.model.GUIBackend.canSend(GUIBackend.java:662)
	at com.willwinder.ugs.nbp.core.actions.StartAction.isEnabled(StartAction.java:75)
	at com.willwinder.ugs.nbp.core.actions.StartAction.lambda$UGSEvent$0(StartAction.java:70)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.desktop/java.awt.EventQueue$4.run(Unknown Source)
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/java.awt.EventQueue.dispatchEvent(Unknown Source)
[catch] at org.netbeans.core.TimableEventQueue.dispatchEvent(TimableEventQueue.java:136)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.desktop/java.awt.EventDispatchThread.run(Unknown Source)
SEVERE [null]: Last record repeated more than 10 times, further logs of this record are ignored until the log record changes.
SEVERE [null]: Last record repeated 70 times in total.
SEVERE [com.willwinder.universalgcodesender.GrblController]: Error while processing response <ok>: An unexpected command was completed by the controller.
com.willwinder.universalgcodesender.AbstractController$UnexpectedCommand: An unexpected command was completed by the controller.
	at com.willwinder.universalgcodesender.AbstractController.commandComplete(AbstractController.java:802)
[catch] at com.willwinder.universalgcodesender.GrblController.rawResponseHandler(GrblController.java:152)
	at com.willwinder.universalgcodesender.AbstractController.rawResponseListener(AbstractController.java:824)
	at com.willwinder.universalgcodesender.AbstractCommunicator.sendEventToListeners(AbstractCommunicator.java:172)
	at com.willwinder.universalgcodesender.AbstractCommunicator.lambda$new$0(AbstractCommunicator.java:194)
	at java.base/java.lang.Thread.run(Unknown Source)
WARNING [org.openide.util.ImageUtilities]: Initial slashes in Utilities.loadImage deprecated (cf. #20072): /org/netbeans/core/windows/resources/error.png
INFO [com.willwinder.ugs.nbp.core.actions.ConnectDisconnectAction]: openclose button, connection open: true
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Disconnecting.

breiler, 2022-01-20:
@c0usin can you post the log from the console with the response from the controller? Which firmware version of GRBL do you have installed?

I'm guessing the problem could be some of the following:
1. Old version of GRBL (like in the original post) - Upgrade to a more recent version of GRBL
2. If the output from the controller contains garbage characters or is missing something in the status report - Check the USB cable, it is likely caused by EMI or a bad controller
3. Do you have other peripherals communicating with the controller at the same time (such as a MPG) - this won't work with UGS and you need to disconnect it.

c0usin, 2022-01-21:
Thank You for your reply @breiler .

1. GRBL version is 1.1
2. The output seems to me it doesn't have any wrong character, for this I checked the Verbose output and copied that here.
3. Only UGS was comunicating.

Also tried with LaserGRBL, and there I didn't get this error.

The log from console:
**** Connected to COM3 @ 115200 baud ****

Grbl 1.1e ['$' for help]
>>> $$
$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 7    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 1    (Hard limits enable, boolean)
$22 = 0    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 411.000    (X-axis travel resolution, step/mm)
$101 = 417.000    (Y-axis travel resolution, step/mm)
$102 = 411.000    (Z-axis travel resolution, step/mm)
$110 = 500.000    (X-axis maximum rate, mm/min)
$111 = 500.000    (Y-axis maximum rate, mm/min)
$112 = 500.000    (Z-axis maximum rate, mm/min)
$120 = 10.000    (X-axis acceleration, mm/sec^2)
$121 = 10.000    (Y-axis acceleration, mm/sec^2)
$122 = 10.000    (Z-axis acceleration, mm/sec^2)
$130 = 200.000    (X-axis maximum travel, millimeters)
$131 = 200.000    (Y-axis maximum travel, millimeters)
$132 = 200.000    (Z-axis maximum travel, millimeters)
ok
>>> $G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
[HLP:$$ $# $G $I $N $x=val $Nx=line $J=line $SLP $C $X $H ~ ! ? ctrl-x]
[Error] Error while processing response <ok>: An unexpected command was completed by the controller.
[HLP:$$ $# $G $I $N $x=val $Nx=line $J=line $SLP $C $X $H ~ ! ? ctrl-x]
[Error] Error while processing response <ok>: An unexpected command was completed by the controller.

In this log now I'm seeing some characters exactly after connection made, don't know.

breiler, 2022-01-21:
I don't think that the characters in the beginning should cause any problems. What confuses me are these responses from the controller:
```
[HLP:$$ $# $G $I $N $x=val $Nx=line $J=line $SLP $C $X $H ~ ! ? ctrl-x]
[Error] Error while processing response : An unexpected command was completed by the controller.
[HLP:$$ $# $G $I $N $x=val $Nx=line $J=line $SLP $C $X $H ~ ! ? ctrl-x]
[Error] Error while processing response : An unexpected command was completed by the controller.
```

That response is sent from the controller when the command `$` is sent, which UGS never does. So I have no idea what is going on... Btw what is the reason you are running the older `Grbl 1.1e`? I assume it is from [gnea/grbl](https://github.com/gnea/grbl/releases) and is the oldest available releases.

>Also tried with LaserGRBL, and there I didn't get this error.

This is a UGS specific error, I don't know how LaserGRBL handles responses it can't understand. But in UGS we use a more strict approach to handle the protocol to make sure both UGS and the controller is having the same state.

c0usin, 2022-01-21:
I got this GRBL with my newly bought MKS DLC v2.1, and this firmware is in hex file

Would I try to upload the GRBL firmware from github?

breiler, 2022-01-21:
@c0usin I don't know what I am supposed to do with this. It is impossible for us to support a firmware that we don't know what it is. The manufacturer is more than welcome to contact me to resolve this, but without the actual firmware or hardware I will not be able to reproduce the error and fix it.

I read about that these kinds of problems could be caused by the controller if using a TFT-screen, but this is just wild guesses in what's going on:
https://forum.v1engineering.com/t/grbl-and-tft-screens-and-some-wifi-too/24786/6

c0usin, 2022-01-21:
Updated the firmware with new version 1.1h from here: https://github.com/gnea/grbl/releases
The problem already appears.

Disconnected the TFT screen resolved this problem. Interesting.

Thank you, maybe also I need to update the firmware on the TFT.

breiler, 2022-01-21:
@c0usin great, let us know how the TFT firmware upgrade works for you. Do you have a link to the purchased hardware?

c0usin, 2022-01-21:
Ok, I will let you know how will work. I found the original Makerbase here in git and I can get the firmware from there.

c0usin, 2022-01-21:
Updated the firmware of TFT to latest (MKS_TFT24_CNC_V1.0.1), problem remains.
I will try to add a wifi connection, maybe that will help because as I see no possibility to set without wifi. In settings appear disconnected but in config wifi modul is connected, probably this is the problem.

c0usin, 2022-02-02:
As I said before, I let You know if I found soemthing.
Also bought an MKS DLC32 with TFT35, just arrived yesterday and I installed to my CNC/Laser.
The UGS program doesn't gives any error when TFT is connected to the new board.

So I think there is an error not in the software of MKS DLC V2.1 (this is unsing GRBL 1.1h) otherwise in the software of TFT2.4 for this board.

inventioFortunata, 2025-04-26:
For me it was the Stop button, which I have accidentally pressed. I had to rotate and pull it out and then everything worked like a charm.