winder/Universal-G-Code-Sender #1492: Problem connecting to CNC via GRBL on Win 10 Pro system

loprestibj, 2020-12-13:
**Describe the bug**
I have a PC running Windows 10 Professional and I cannot connect to my Millright Mega V CNC router using UGS/GRBL.   I have tried damn near everything I can think of.   Updated Windows, updated the BIOS, all system drivers,  re-installed Arduino drivers, re-installed Java, re-installed UGS, plugged and un-plugged the USB cable.   

**To reproduce**
Steps to reproduce the behavior:
1. When I initialize UGS, it sometimes connects and sometimes does not.   When it connects, I get the message that UGS is connected to COM3 at 115200 baud.  But, the machine status remains unknown.  

2. When I try to get the state of the machine or do a soft reset I get the "No supported view parser" message or the "GRBL has not finished loading message".  

3.   I have UGS installed on a Mac, and it works perfectly fine with the same machine, USB cable, and controller.   I also have an older Win 7 PC, and I have installed UGS on that machine and it also works as expected.  

**Expected behavior**
When I refresh the port and connect, I expect UGS to find the machine and initialize GRBL properly.   
I expect to be able to communicate and control the machine, run g-code, etc. 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version**
Current UGS platform version (downloaded today)

**Hardware**
Millright Mega V controlled by Arduino UNO w/GRBL

**Operating system (please complete the following information):**
Windows 10 Professional

**Additional context**
Seems related to the USB, but I tried switching USB 2.0 ports with no success.   Some of the USB ports don't recognize the Arduino controller when you plug it in while others do.   The Device Manager indicates that the device is working properly.

breiler, 2020-12-13:
We have got several reports on this over the last week, so this is likely a new problem.

Could you check that the connection driver is JSerialComm, this is done in the settings here: 
[image]

Could there be a antivirus program causing the problem?

loprestibj, 2020-12-13:
Yes- JSerialComm is the connection driver.   Antivirus software is disabled.   Firewalls are all shut off.   I thought it might be something in Windows Security so I disabled everything I could-- still not working.   Interestingly, I also have Inventables Easel Pro, which also does not connect now.   Both Easel and UGS were working fine on this PC and then one day in the last week or two they weren't.   Seems to be issue between GRBL and USB.   I'll bet it is some Windows automatic update that fouled things up.   It does not seem specific to UGS, as Easel has the problem also... more at the device level.  

Let me know how I can help to sort this out.   I love UGS!  

Thanks
Brian

loprestibj, 2020-12-15:
I did a clean install of Windows 10 Pro and reinstalled Arduino, Java, UGS... and it worked again for an afternoon.   Now I have the same problem again and also some new ones-  driver error code 10:  the device cannot start.  Reinstalled Arduino, rebooted machine, still flags a driver error.

breiler, 2020-12-15:
It does sound more like a Windows issue and not a UGS issue as you mentioned. I found this page with different steps for troubleshooting:
https://windowsreport.com/arduino-problem-windows-10/

I will see if I can get a Windows machine to debug this as well

loprestibj, 2020-12-15:
Ok-  will go through the list to see where things are going wrong.   

Does UGS write out a log file that is more informative than the output to the console?

breiler, 2020-12-15:
Yes, have a look at this:
https://github.com/winder/Universal-G-Code-Sender/wiki/Configuration#log-files

loprestibj, 2020-12-15:
Here's the logfile:

-------------------------------------------------------------------------------
>Log Session: Tuesday, December 15, 2020 at 9:41:52 AM Eastern Standard Time
>System Info: 
  Product Version         = Universal Gcode Platform 12.0-631bd69cd6112b1cc4c892c24e3e605b1ba04241
  Operating System        = Windows 10 version 10.0 running on amd64
  Java; VM; Vendor        = 13.0.1; OpenJDK 64-Bit Server VM 13.0.1+9; AdoptOpenJDK
  Runtime                 = OpenJDK Runtime Environment 13.0.1+9
  Java Home               = C:\Users\lopre\Downloads\ugs-platform-app-win\ugsplatform-win\jdk\jdk-13.0.1+9-jre
  System Locale; Encoding = en_US (ugsplatform); Cp1252
  Home Directory          = C:\Users\lopre
  Current Directory       = C:\Users\lopre\Downloads\ugs-platform-app-win\ugsplatform-win
  User Directory          = C:\Users\lopre\.ugsplatform\2.0.7-SNAPSHOT\dev
  Cache Directory         = C:\Users\lopre\.ugsplatform\2.0.7-SNAPSHOT\dev\var\cache
  Installation            = C:\Users\lopre\Downloads\ugs-platform-app-win\ugsplatform-win\etc
                            C:\Users\lopre\Downloads\ugs-platform-app-win\ugsplatform-win\ide
                            C:\Users\lopre\Downloads\ugs-platform-app-win\ugsplatform-win\bin
                            C:\Users\lopre\Downloads\ugs-platform-app-win\ugsplatform-win\ugsplatform
                            C:\Users\lopre\Downloads\ugs-platform-app-win\ugsplatform-win\extra
                            C:\Users\lopre\Downloads\ugs-platform-app-win\ugsplatform-win\java
                            C:\Users\lopre\Downloads\ugs-platform-app-win\ugsplatform-win\platform
                            C:\Users\lopre\Downloads\ugs-platform-app-win\ugsplatform-win\platform
  Boot & Ext. Classpath   = 
  Application Classpath   = C:\Users\lopre\Downloads\ugs-platform-app-win\ugsplatform-win\platform\lib\boot.jar;C:\Users\lopre\Downloads\ugs-platform-app-win\ugsplatform-win\platform\lib\org-openide-modules.jar;C:\Users\lopre\Downloads\ugs-platform-app-win\ugsplatform-win\platform\lib\org-openide-util-lookup.jar;C:\Users\lopre\Downloads\ugs-platform-app-win\ugsplatform-win\platform\lib\org-openide-util-ui.jar;C:\Users\lopre\Downloads\ugs-platform-app-win\ugsplatform-win\platform\lib\org-openide-util.jar
  Startup Classpath       = C:\Users\lopre\Downloads\ugs-platform-app-win\ugsplatform-win\platform\core\asm-7.2.jar;C:\Users\lopre\Downloads\ugs-platform-app-win\ugsplatform-win\platform\core\asm-commons-7.2.jar;C:\Users\lopre\Downloads\ugs-platform-app-win\ugsplatform-win\platform\core\asm-tree-7.2.jar;C:\Users\lopre\Downloads\ugs-platform-app-win\ugsplatform-win\platform\core\core-base.jar;C:\Users\lopre\Downloads\ugs-platform-app-win\ugsplatform-win\platform\core\core.jar;C:\Users\lopre\Downloads\ugs-platform-app-win\ugsplatform-win\platform\core\org-netbeans-libs-asm.jar;C:\Users\lopre\Downloads\ugs-platform-app-win\ugsplatform-win\platform\core\org-openide-filesystems-compat8.jar;C:\Users\lopre\Downloads\ugs-platform-app-win\ugsplatform-win\platform\core\org-openide-filesystems.jar;C:\Users\lopre\Downloads\ugs-platform-app-win\ugsplatform-win\ugsplatform\core\locale\core_ugsplatform.jar
-------------------------------------------------------------------------------
INFO [com.willwinder.ugs.nbp.core.lifecycle.startup]: Loading LocalizingService...
INFO [com.willwinder.universalgcodesender.utils.SettingsFactory]: Log location: C:\Users\lopre\.ugs\UniversalGcodeSender.json
INFO [com.willwinder.universalgcodesender.utils.SettingsFactory]: Loading settings.
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Applying settings.
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Applying new command processor
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding UGSEvent listener: RunFromService
WARNING [ActionRegistrationService]: Could not load class for action Actions/Machine/com.willwinder.ugs.nbp.core.services.JogActionService.JogSizeAction.decrease.feed.instance
WARNING [ActionRegistrationService]: Could not load class for action Actions/Machine/com.willwinder.ugs.nbp.core.services.JogActionService.JogSizeAction.decrease.instance
WARNING [ActionRegistrationService]: Could not load class for action Actions/Machine/com.willwinder.ugs.nbp.core.services.JogActionService.JogSizeAction.decrease.z.instance
WARNING [ActionRegistrationService]: Could not load class for action Actions/Machine/com.willwinder.ugs.nbp.core.services.JogActionService.JogSizeAction.divide.feed.instance
WARNING [ActionRegistrationService]: Could not load class for action Actions/Machine/com.willwinder.ugs.nbp.core.services.JogActionService.JogSizeAction.divide.instance
WARNING [ActionRegistrationService]: Could not load class for action Actions/Machine/com.willwinder.ugs.nbp.core.services.JogActionService.JogSizeAction.divide.z.instance
WARNING [ActionRegistrationService]: Could not load class for action Actions/Machine/com.willwinder.ugs.nbp.core.services.JogActionService.JogSizeAction.inch.instance
WARNING [ActionRegistrationService]: Could not load class for action Actions/Machine/com.willwinder.ugs.nbp.core.services.JogActionService.JogSizeAction.increase.feed.instance
WARNING [ActionRegistrationService]: Could not load class for action Actions/Machine/com.willwinder.ugs.nbp.core.services.JogActionService.JogSizeAction.increase.instance
[... cut ...]
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding UGSEvent listener: OverridesPanel
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding controller listener: OverridesPanel
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding UGSEvent listener: ActionButtonPanel
INFO [com.willwinder.universalgcodesender.utils.SettingsFactory]: Saving settings.
INFO [org.netbeans.ui.indexing]: Indexing started, time from last indexing 0 ms.
INFO [org.netbeans.ui.indexing]: Indexing finished, indexing took 0 ms.

loprestibj, 2020-12-15:
And now the Arduino device is signaling a driver error

loprestibj, 2020-12-15:
i should add that UGS works fine on Mac and also Windows 7 box.   The problem does not seem to be related to Arduino hardware or sketch,   There is probably some Windows 10 "feature" that confounds unsigned applications from starting some critical service.  I disabled driver signing enforcement-- still won't run on Win 10 Pro.

loprestibj, 2020-12-15:
actually, now I am getting the exact same problem on Windows 7 machine !!!

loprestibj, 2020-12-17:
The log file shows a FirmwareSettingsException: 

com.willwinder.universalgcodesender.firmware.FirmwareSettingsException: Couldn't find setting with key: $22

breiler, 2020-12-18:
Looking through the log file it looks like its sending the `$$` command which would mean that it gets the GRBL welcome string. So it manages to connect to the controller. The log file message that it can't find the setting indicates that it didn't receive a proper response on the `$$`-command. 

Could you post the console log from a clean connection with the machine?

I'm wondering this could be the result from a bad USB-cable, do you have any other to test with?

loprestibj, 2020-12-18:
Yes-  it didn't really change anything

i have learned a bit more about this condition.  Interestingly, I can't communicate with the board through the Arduino IDE serial monitor.   I discovered that I can sometimes 

get the machine to connect briefly by unplugging and plugging the usb cable back in to the same port.  UGS will flag some errors and the machine will disconnect.  You cannot connect again until you unplug and plug the machine back in.     

Screenshots of the error messages and a logfile from a successful connection are attached and pasted below.   

Thank you sincerely for the help!

[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/5716394/messages.log)
[image]
[image]

-------------------------------------------------------------------------------
>Log Session: Friday, December 18, 2020 at 10:00:52 AM Eastern Standard Time
>System Info: 
  Product Version         = Universal Gcode Platform 12.0-631bd69cd6112b1cc4c892c24e3e605b1ba04241
  Operating System        = Windows 10 version 10.0 running on amd64
  Java; VM; Vendor        = 13.0.1; OpenJDK 64-Bit Server VM 13.0.1+9; AdoptOpenJDK
  Runtime                 = OpenJDK Runtime Environment 13.0.1+9
  Java Home               = C:\Users\lopre\OneDrive\Documents\ugs-platform-app-win\ugsplatform-win\jdk\jdk-13.0.1+9-jre
  System Locale; Encoding = en_US (ugsplatform); Cp1252
  Home Directory          = C:\Users\lopre
  Current Directory       = C:\Users\lopre\OneDrive\Documents\ugs-platform-app-win\ugsplatform-win
  User Directory          = C:\Users\lopre\.ugsplatform\2.0.7-SNAPSHOT\dev
  Cache Directory         = C:\Users\lopre\.ugsplatform\2.0.7-SNAPSHOT\dev\var\cache
  Installation            = C:\Users\lopre\OneDrive\Documents\ugs-platform-app-win\ugsplatform-win\etc
                            C:\Users\lopre\OneDrive\Documents\ugs-platform-app-win\ugsplatform-win\ide
                            C:\Users\lopre\OneDrive\Documents\ugs-platform-app-win\ugsplatform-win\bin
                            C:\Users\lopre\OneDrive\Documents\ugs-platform-app-win\ugsplatform-win\ugsplatform
                            C:\Users\lopre\OneDrive\Documents\ugs-platform-app-win\ugsplatform-win\extra
                            C:\Users\lopre\OneDrive\Documents\ugs-platform-app-win\ugsplatform-win\java
                            C:\Users\lopre\OneDrive\Documents\ugs-platform-app-win\ugsplatform-win\platform
                            C:\Users\lopre\OneDrive\Documents\ugs-platform-app-win\ugsplatform-win\platform
  Boot & Ext. Classpath   = 
  Application Classpath   = C:\Users\lopre\OneDrive\Documents\ugs-platform-app-win\ugsplatform-win\platform\lib\boot.jar;C:\Users\lopre\OneDrive\Documents\ugs-platform-app-win\ugsplatform-win\platform\lib\org-openide-modules.jar;C:\Users\lopre\OneDrive\Documents\ugs-platform-app-win\ugsplatform-win\platform\lib\org-openide-util-lookup.jar;C:\Users\lopre\OneDrive\Documents\ugs-platform-app-win\ugsplatform-win\platform\lib\org-openide-util-ui.jar;C:\Users\lopre\OneDrive\Documents\ugs-platform-app-win\ugsplatform-win\platform\lib\org-openide-util.jar
  Startup Classpath       = C:\Users\lopre\OneDrive\Documents\ugs-platform-app-win\ugsplatform-win\platform\core\asm-7.2.jar;C:\Users\lopre\OneDrive\Documents\ugs-platform-app-win\ugsplatform-win\platform\core\asm-commons-7.2.jar;C:\Users\lopre\OneDrive\Documents\ugs-platform-app-win\ugsplatform-win\platform\core\asm-tree-7.2.jar;C:\Users\lopre\OneDrive\Documents\ugs-platform-app-win\ugsplatform-win\platform\core\core-base.jar;C:\Users\lopre\OneDrive\Documents\ugs-platform-app-win\ugsplatform-win\platform\core\core.jar;C:\Users\lopre\OneDrive\Documents\ugs-platform-app-win\ugsplatform-win\platform\core\org-netbeans-libs-asm.jar;C:\Users\lopre\OneDrive\Documents\ugs-platform-app-win\ugsplatform-win\platform\core\org-openide-filesystems-compat8.jar;C:\Users\lopre\OneDrive\Documents\ugs-platform-app-win\ugsplatform-win\platform\core\org-openide-filesystems.jar;C:\Users\lopre\OneDrive\Documents\ugs-platform-app-win\ugsplatform-win\ugsplatform\core\locale\core_ugsplatform.jar
-------------------------------------------------------------------------------
INFO [com.willwinder.ugs.nbp.core.lifecycle.startup]: Loading LocalizingService...
INFO [com.willwinder.universalgcodesender.utils.SettingsFactory]: Log location: C:\Users\lopre\.ugs\UniversalGcodeSender.json
INFO [com.willwinder.universalgcodesender.utils.SettingsFactory]: Loading settings.
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Applying settings.
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Applying new command processor
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Adding UGSEvent listener: RunFromService
WARNING [ActionRegistrationService]: Could not load class for action Actions/Machine/com.willwinder.ugs.nbp.core.services.JogActionService.JogSizeAction.decrease.feed.instance
WARNING [ActionRegistrationService]: Could not load class for action Actions/Machine/com.willwinder.ugs.nbp.core.services.JogActionService.JogSizeAction.decrease.instance
WARNING [ActionRegistrationService]: Could not load class for action Actions/Machine/com.willwinder.ugs.nbp.core.services.JogActionService.JogSizeAction.decrease.z.instance
WARNING [ActionRegistrationService]: Could not load class for action Actions/Machine/com.willwinder.ugs.nbp.core.services.JogActionService.JogSizeAction.divide.feed.instance
[... cut ...]
	at com.willwinder.universalgcodesender.AbstractCommunicator.lambda$new$0(AbstractCommunicator.java:194)
	at java.base/java.lang.Thread.run(Unknown Source)
INFO [org.netbeans.core.ui.focus]: External Changes Refresh on focus gain disabled
WARNING [com.willwinder.universalgcodesender.gcode.util.GcodeParserUtils]: An unknown gcode command was detected in: [GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0
INFO [com.willwinder.ugs.nbp.core.actions.ConnectDisconnectAction]: openclose button, connection open: true
INFO [com.willwinder.universalgcodesender.model.GUIBackend]: Disconnecting.

wile1411, 2021-10-31:
Would it help to have a log of the exact same issue from a Ubuntu install?

breiler, 2021-11-01:
This should have been fixed in #1637 and is available in version 2.0.8.

closed 2021-11-30 by breiler

WalterHynson, 2023-07-13:
well it looks like its back again,fails to connect(connecting all day)

breiler, 2023-07-13:
@WalterHynson please post a new issue with details of your installation.

PhantomStaubli, 2023-07-23:
[breiler](https://github.com/breiler) I am unable to connect Mac OS to my usual controller as of today. Please advise.

PhantomStaubli, 2023-07-23:
[image]

breiler, 2023-07-23:
@PhantomStaubli please create new issue with all details. You are connecting to the controller through Bluetooth. Is this correct?