winder/Universal-G-Code-Sender #1493: Application doesn't start on Windows 7

Krasnogorov, 2020-12-16:
**Describe the bug**
Application show error and finish work. 

**To reproduce**
Steps to reproduce the behavior:
1. Download version 2.0.7 from official site
2. Unarchive it to disk D
3. Go to bin folder
4. Launch ugsplatform.exe
4. See error

**Expected behavior**
Application should start.

**Screenshots**
Screenshot is here [https://www.dropbox.com/s/eqlz9nrsgnway4p/Image%201.png?dl=0](url)
Text on screenshot :
"Failed to create process Version "%1" is not supported with Windows version, which work on this computer. Check information about system to know which program's version you need x86(32bit) or x64(64bit) and receive it from software provider."

**Version**
 UGS Platform 2.0.7

**Hardware**
GRBL on Arduino UNO

**Operating system (please complete the following information):**
Window 7 32-bit

**Additional context**
CPU : Celeron Dual-core T3300 @2GHz 2,81 Ghz
RAM: 4 GB
Path to launch : D:\ugsplatform-win\ugsplatform-win\bin\ugsplatform.exe
Version 1.0.9 is working normally on this PC.

If I launch program (ugsplatform64.exe) from this archive on other PC (Win10/64bit) it works without any problem.

breiler, 2021-01-04:
Sorry, we are packaging UGS using a 64-bit JVM.

You could try to download this Java-version: https://github.com/AdoptOpenJDK/openjdk13-binaries/releases/download/jdk-13.0.1%2B9/OpenJDK13U-jre_x86-32_windows_hotspot_13.0.1_9.zip

Remove the directory under `ugsplatform-win/jdk/jdk-13.0.1+9-jre` with the one from this zip.

breiler, 2021-01-04:
@winder how would you feel about replacing the 64-bit JVM with a 32-bit by default in the Windows package?
This would make things simpler for people with older 32-bit machines using Win7 and Win8.

breiler, 2021-01-06:
@Krasnogorov I have created a new package with a bundled 32-bit JRE. I would appreciate if you could give it a try:
https://drive.google.com/file/d/1eiQJca-CD5_xl59NR2fbtXZpoJeArZjf/view?usp=sharing

closed 2021-01-07 by breiler

SerhiiKrasnohorov, 2021-01-08:
> @Krasnogorov I have created a new package with a bundled 32-bit JRE. I would appreciate if you could give it a try:
> https://drive.google.com/file/d/1eiQJca-CD5_xl59NR2fbtXZpoJeArZjf/view?usp=sharing

Yes. It works for me. Thank you!

PR #1506 (2021-01-07): Now using a 32-bit JRE for Windows and JRE8 for raspberry pi.
Changed the default JRE:s for Windows and RaspberryPi.
Fixes #1493 
Fixes #1505