winder/Universal-G-Code-Sender #1496: Universal Gcode Platform 20201202 - no file loaded

owenduffy, 2020-12-24:
**Describe the bug**
I click FILE/OPEN and select and nc file that has record length of 58 bytes that causes an error.

[image]

Nothing shows in the visualiser.

The record is from Freecad.

**To reproduce**
Steps to reproduce the behavior:
As above.

**Expected behavior**
I expected the file to load, and the path to show in the visualiser.

**Screenshots**
Above.

**Version**
UGS Platform 2.0.7

**Hardware**
e.g. GRBL-esp32

**Operating system (please complete the following information):**
Windows 10

**Additional context**
This works ok in nightly 2.0 JUL 14 2020.

As I understand it, the record sent to the controller may be shorter or longer than the input record due to certain processing in UGS, so I cannot understand why it imposes this limit at the input end of things?

BTW, as I understand it, grbl-esp32 allows input records up to 128B.

Owen

breiler, 2020-12-30:
Earlier versions of GRBL was sensitive to long lines. There is a command processor in UGS that checks this, making sure we are not overfilling any buffers. The default value is 70 characters, but you might be running an older configuration.

You can simply disable this command processor or change its value here under `Command Length Processor`:
[image]

owenduffy, 2020-12-30:
Thanks.

As I stated, I am running 2.07 which I downloaded days before reporting this issue. The nightly had the same problem.

[image]

I had already set it CommandLengthProcessor to 128 before observing and reporting the issue.

I am using grbl_esp32 which I understand has a 128B buffer.

Owen

breiler, 2020-12-30:
Thanks for the feedback.

It was the value `50` that puzzled me which lead me to believe it was an old setting value from a previous version. 
But I think I've found the problem, if there is any error while loading the preprocessors it will default to the value 50.

Could you upload the log file: https://github.com/winder/Universal-G-Code-Sender/wiki/Configuration#log-files

owenduffy, 2020-12-30:
Ok, the log files are actually at C:\Users\owen\.ugsplatform\2.0.7-SNAPSHOT\dev\var\log (not as stated in the Wiki).

[messages.zip](https://github.com/winder/Universal-G-Code-Sender/files/5753718/messages.zip)

breiler, 2020-12-30:
That log didn't contain log rows from when you opened the file. Could you retry opening a connection to GRBL and then open a file.

owenduffy, 2020-12-30:
Ok, apologies, I misunderstood... I thought you just want package startup.

This problem occurs if I try to open a file prior to connecting to grbl_esp32. If it is not valid to do that, the file open option should be disabled.

For this test, the command length processor is disabled (or supposed to be).

[messages.zip](https://github.com/winder/Universal-G-Code-Sender/files/5755084/messages.zip)

Owen

breiler, 2020-12-30:
Ahh, ok when it is not connected to the machine the default preprocessors are loaded which is limited to 50 characters. Such a default rule is unnecessary so I'm going to remove it.

But you are not getting this error when loading a file after you connected to the machine, right?

owenduffy, 2020-12-30:
I will have to go down to the shed to try, it is 4am, so it will be a few hours.

Did you note that I had disabled the rule, so that is being ignored as well at this time.

I have noticed changing behavior here. I did like the file reload button but it disappeared. A nice feature would be if UGS recognised that a file had been updated and prompted to reopen it.

I have been using a 14 Jul nightly, and it shows the error in the bottom of the window, but it loads and displays the file ok (again prior to connection to grbl).

Since grbl has a larger command buffer, perhaps it is deserving of its own machine type in your list.

I will report back in a few hours.

Owen

breiler, 2020-12-30:
Yes, I noticed that you had the preprocessor disabled for GRBL. But it only loads those settings if it is connected to GRBL, otherwise it will load the default settings which are set a bit low.

I'm planning some changes to better detect the variant of GRBL it is connected to for enabling extended features in grbl_esp32 and grblHAL. I'm hoping we can add these extensions without introducing other machine types in the dropdown to make it easier for our users. I have not considered the CommandLengthProcessor though, but I'm wondering if it is likely that programs will generate longer lines than 70 characters? I have not seen this problem before so I'm thinking that these changes can wait.

Regarding the file change, this would be a nice feature. I will create an feature request for this!

owenduffy, 2020-12-30:
Ok.

This gets complicated and messy?

I am getting the picture that when a file is opened, it is only correctly processed if the grbl connection is established? If this is true, this is a trap unless UGS reprocesses the input file (as it was) after opening grbl?

[log.zip](https://github.com/winder/Universal-G-Code-Sender/files/5755453/log.zip)

So, with grbl_esp32 connected, it did load... but a different error in the status line, see the screenshot in the zip. It complains of multiple speed command on one line, I can see only one speed command, and there is only one, but it uses a floating point number and perhaps UGS splits that at the dot?

Owen

breiler, 2020-12-30:
It will reprocess the file after connecting to any controller using the preprocessor settings for that controller.

The file loaded for me without any problems. I even tried inactivating the command length processor.

I can't explain the error you got. I remember that there was a similar problem about a year ago where it tried to parse certain comments as gcode, but this was fixed.

owenduffy, 2020-12-30:
Shall I get a nightly build when you have applied your changes, and test that?

breiler, 2020-12-30:
That will only fix the problem with loading gcode files when not being connected. If you load the file after you have connected it will not make a difference.

owenduffy, 2020-12-30:
I understand that, and there is the speed command problem I see and you don't... though it seems to not prevent the file loading.

owenduffy, 2020-12-31:
Thanks... Owen.

owenduffy, 2021-01-03:
Tested on 20210103 nightly and the reported problem is fixed, thanks... Owen

PR #1503 (2021-01-03): Removed the CommandLengthProcessor from the default rules.
This caused problems when loading files while not connected to the machine.
Fixes #1496
closed 2021-01-03 by owenduffy