winder/Universal-G-Code-Sender #1497: Bug: Limit Switches X and Y axis swapped

BentliJ, 2020-12-26:
From the "Tool Wizard "  -> Limit Switches.   When The Y limit switches is triggered the Tool wizard shows that X limit switch is triggered. When the X limit switch is triggered then the Tool Wizard shows Y switch.   We cannot fix in the hardware because then other apps that display them correctly will be Faulty.   Small Code Fix  - would appreciate a  Fix.   The Homing Functionality should then also work. Version UGS 20201220.

breiler, 2020-12-29:
Which apps displays the limits correctly?

We are parsing the limits reported by GRBL as XYZ and displays them as is. I've double checked the code and it looks like it is doing what it supposed to.

What version of GRBL are you using and what settings are you using?

BentliJ, 2020-12-29:
Hi , 

 

Using GRBL 1.1h.

 

Oh,  Oh,

 

My humblest apologies!!!

 

I swapped 2 wires in my little distribution box on my CNC machine.

 

Rechecked with UGS the plain Java one.

The verbose mode now show X.

 

Problem Solved.  

 

So, so sorry for wasting  our  time.  K

 

Many Thanks.

 

Do you perhaps know if the GRBL homing works on 1.1h ?

breiler, 2020-12-29:
No worries!

The homing process should work fine on GRBL 1.1h. I have struggled a lot with this as EMI falsly triggers the limit pins. But that has nothing to do with the software, just my bad hardware skills. ;)

BentliJ, 2020-12-29:
Got the Homing to work.

 

I had the same problem.  It is NOT EMF.

 

Every time I issued a M3 command  to turn on the spindle  the limits switches triggered by itself.

Since I had  LED’s on the switches  I could see they are NOT.

 

I modified the GRBL code to display the Limit status for me.

It showed a 0 – None  - No limit switch were triggered, yet GRBL said it is so.

It is a little Bug in the Firmware in the ISR routine in limits.c

 

My M3 command and M5 work like a dream.

 

If you are interested I will send you the code.

Literally  less than 5 minutes to implement.

 

Best regards.

closed 2020-12-29 by breiler