winder/Universal-G-Code-Sender #1498: UGS 2.0.7 SNAPSHOT driving GRBL gives error20 when sending T1M6

tkms002, 2020-12-27:
My UGS running computer died so I had to get a new computer and had to load the latest UGS on it. 
My machine runs GRBL 1.1.
I think I have all the modifications applied to it that I had to my old UGS.
The machine moves and the probe module is working.
When I try to run a GCode file I get the following error.
{ERROR} An error was detected while sending 'T1M6'. (error:20)
The version I was running before must have just ignored this as all my gcode files have it in.
Am I missing something or do I need to change something?
Thanks

Punmageddon, 2020-12-27:
I've been using UGS for a while to send Gcode created in Fusion to my GRBL cnc. I've always resorted to deleting tool change commands ("T1M6"). Pretty sure this is default behavior. UGS usually doesn't ignore errors that GRBL reports, but throws them back at the user.

tkms002, 2020-12-27:
Well the release I was running on my old computer must have ignored them because I never got this error before.
I set up Fusion to not generate them and all is well but I will have to regenerate all my older gcode files.

AndyCXL, 2021-07-20:
You must have had the 'remover' pattern processor setup (preferences->-ugs->-controller options menu) and removing tool change commands. If you can't remember what this setting is or was, something like " Td+M6 " as the regex pattern will find and remove these

breiler, 2022-01-14:
The editor will now highlight problems like this. When UGS now encounters M6 it will pause the stream and show an error allowing you to continue.

closed 2022-01-14 by breiler