winder/Universal-G-Code-Sender #1499: Hotkey for console input field

Punmageddon, 2020-12-27:
**Problem statement**
I enjoy using key bindings to for example quickly stop the machine without having to hit the power switch.
I would like to be able to send custom gcode snippets quickly. I could not find a keybinding to focus the input field of the console window.

**Suggestion**
I would like one binding setting to focus the console input field for input.
Example usage:
1. Press \<my new hotkey\>
2. Typing \<my gcode\>
3. Press Enter (sends to machine)

**Alternative suggestion**
Since I often use the same snippet, I considered using macros. Sadly they can also not be invoked using key combinations.

**Additional context**
Using the mouse is slow and cumbersome (especially on a laptop), which is why I would like to use the keyboard for everything I do in UGS. I am also considering building a jog pendant that emulates a usb keyboard, sadly this means its functionality is bound to what key bindings are supported. Being able to inject gcode snippets would allow me to implement essentially unlimited functionality into the pendant.

breiler, 2020-12-30:
Thanks for a good feature request!

I created a pull request to fix this. When this is merged you will be able to add a shortcut to the console window:
[image]

The macros should also be able to map to shortcuts, look for the category "Macros" in the keymap settings:
[image]

If you know some coding you could also use the REST-API to access your machine and automate things. There is little documentation on this, but if you look at the REST-endpoints it should be pretty straight forward:
https://github.com/winder/Universal-G-Code-Sender/tree/master/ugs-pendant/src/main/java/com/willwinder/universalgcodesender/pendantui/v1/resources

An example, if you do a GET on `http://localhost:8080/api/v1/macros/getMacroList`
```

[
  {"gcode":"G91 X0 Y0;", "description":null, "name":"Go to Zero"},
  {"gcode":"$1", "description":null, "name":"Macro #1"}
]
```

closed 2021-01-03 by breiler

owenduffy, 2021-01-03:
This will be really usefull, thanks, Owen.

PR #1502 (2021-01-03): Now selects command field when component is selected
Now selects command field when component is selected

Made it possible to bind a shortcut to the console window and automatically select the command field.

Fixes #1499