winder/Universal-G-Code-Sender #1500: "Jog Here" Command not working - UGS - BobsCNC Evo4

bdrayer21, 2020-12-28:
**Describe the bug**
When I right mouse click in the visualizer and go to "coordinates" and then select "jog here" I get an error message.

Error Message: 
An error was detected while sending '$J=G20G90G1X______Y______': (error:16) Jog command has no '=' or contains prohibited g-code. Streaming has been paused.

**To reproduce**
Steps to reproduce the behavior:
1. Go to "visualizer"
2. Right mouse click location you would like to view coordinates
3. Scroll down to "coordinates X=xx.xxxx Y=xx.xxxx
4. Click "jog to here" from other pop-up window.
5. Get error

**Expected behavior**
I'd expect that when I select "jog to here" the machine would go to those coordinates. This would assist in setting my z when setting up a project to cut. 

**Screenshots**
[image]
[image]

**Version**
Product Version: Universal Gcode Platform 20190814

**Hardware**
Bobs CNC Arduino Board - GRBL 

**Operating system (please complete the following information):**
 System: Windows 10 version 10.0 running on amd64; Cp1252; en_US (ugsplatform)

**Additional context**
Java: 1.8.0_261; Java HotSpot(TM) 64-Bit Server VM 25.261-b12
Runtime: Java(TM) SE Runtime Environment 1.8.0_261-b12

If I missed any info, please let me know. 

Thanks,
bdrayer21

breiler, 2020-12-29:
Never seen that error. You are running an old version of UGS, could you try if the problem still exists in the latest version 2.0.7

bdrayer21, 2021-01-02:
@breiler - Thanks, I wasn't aware that I didn't have the most up to date version (had tried to "update" from the version of UGS I had, but it said it was all up to date. Just installed 2.0.7 version and the Jog Here command seems to work. Thank you!

closed 2021-01-02 by bdrayer21