winder/Universal-G-Code-Sender #1501: UGS Platform (2.0.7) crashes on macOS Big Sur

voelkerb, 2020-12-29:
Crash on UGS platform startup running under latest macOS Big Sur. 
I know, it's just out for a month or two, but I just wanted to let you know. 
The same goes for classic UGS if you want to visualise the Gcode.
The error msgs points to the need to call all OpenGL function solely from the main thread. 
I guess this was introduced with Big Sur - as the same version works flawlessly on Catalina...

**To reproduce**
UGS platform: startup
UGS classic: load a Gcode file and click "visualize"

**Version**
UGS Platform 2.0.7, UGS Classic 1.0.9

**Hardware**
GRBL initial 1.1

**Operating system**
macOS Big Sur

**Additional context**
Crashed Thread:        28  Java: AWT-EventQueue-0

Exception Type:        EXC_BAD_ACCESS (SIGABRT)
Exception Codes:       EXC_I386_GPFLT
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
-[NSOpenGLContext setView:] must be called from the main thread.

breiler, 2020-12-30:
Thanks for the issue report. The NetBeans project have had a bunch of problems with Big Sur, we will attempt to upgrade to the latest NetBeans once the problems are fixed.

The classic edition is a problem that was introduced in Mojave (I think) and requires a new version 2.3.3 of the OpenGL-library which has not yet been released: https://jogamp.org/jogl/www/

celebie, 2021-03-09:
I am also getting "Universal Gcode Platform can not be opened because the developer can not be verified" error

voelkerb, 2021-03-09:
Hey @celebie,
This is related to the standard macOS Gatekeeper functionality and applies to all Application. 
Just open _System Preferences->Security&Privacy->General_ and allow to open UGS by clicking _Open Anyway_.

voelkerb, 2021-03-09:
The problem seems to be fixed for UGS Platform by the lates Big Sur update (11.2.2). 
Nevertheless, thanks for the support!

closed 2021-03-09 by voelkerb