winder/Universal-G-Code-Sender #1505: UGS does not compile on RPI: 'Node.js' '404 from server'

benclewett, 2021-01-05:
When compelling the latest version on Raspberry PI (Raspberry Pi 3 Model B Plus Rev 1.3, Debian 10.7) I get this error:

./run_platform.sh

ERROR] Failed to execute goal com.github.eirslett:frontend-maven-plugin:1.6:install-node-and-npm (install node and npm) on project ugs-pendant: Could not download Node.js: Got error code 404 from the server. -> [Help 1]

Any help welcome?

breiler, 2021-01-05:
We are using maven to build the node application which can't be built on ARM-based systems. So you need to disable the pendant module from the build:
`mvn install -P '!create-pendant-web-ui'`

If you just want to run UGS on a RasberryPI I would recommend you use the prebuilt image instead of compiling it:
https://github.com/winder/Universal-G-Code-Sender#downloads

benclewett, 2021-01-05:
Joacim,

Thanks for the reply, that does work.  I have version 202001202 installed.

Although at idle the java process sits at exactly 100% cpu, is that
expected?

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+
COMMAND
 1116 pi        20   0  591684 209060  41620 S 101.0  23.4   9:51.36 java

Somebody might also be interested that it starts with this message:

OpenJDK Server VM warning: Options -Xverify:none and -noverify were
deprecated in JDK 13 and will likely be removed in a future release.
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by
org.netbeans.ProxyURLStreamHandlerFactory
(file:/home/pi/ugsplatform-pi/platform/lib/boot.jar) to field
java.net.URL.handler
WARNING: Please consider reporting this to the maintainers of
org.netbeans.ProxyURLStreamHandlerFactory
WARNING: Use --illegal-access=warn to enable warnings of further illegal
reflective access operations
WARNING: All illegal access operations will be denied in a future release

Finally, somebody may want to know that I installed this code based on this
video:  https://www.youtube.com/watch?v=u35L0jGCqFc.   If this method no
longer works, maybe you should put a comment in the script stating this
doesn't work on RPI?  That might stop other users like myself contacting
your support.

However I have it working, so look forward to trying it out!

Regards,

Ben

PS, still sitting on 100% CPU when idle...

Ben

breiler, 2021-01-05:
Could you have a look in the log file:
https://github.com/winder/Universal-G-Code-Sender/wiki/Configuration#log-files

I know there has been problems with the visualizer on RPi and there might be a bunch of errors being spewed in the log.

benclewett, 2021-01-05:
Certainly.  Running from cold we have again 100% CPU:

top - 16:55:30 up 9 min,  3 users,  load average: 1.50, 1.52, 0.79
Tasks: 142 total,   1 running, 141 sleeping,   0 stopped,   0 zombie
%Cpu(s): 25.3 us,  0.3 sy,  0.0 ni, 74.4 id,  0.0 wa,  0.0 hi,  0.0 si,
 0.0 st
MiB Mem :    873.3 total,    265.8 free,    285.2 used,    322.3 buff/cache
MiB Swap:    100.0 total,    100.0 free,      0.0 used.    528.5 avail Mem

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+
COMMAND

 1274 pi        20   0  592696 206360  42116 S 100.7  23.1   7:03.40 java

The new log file from /home/pi/.ugsplatform/2.0.7-SNAPSHOT/dev/var/log is
attached.

I hope you find this useful.

Ben

breiler, 2021-01-05:
Attachments from emails will not be added to github, you would need to add it through the web page.

benclewett, 2021-01-05:
Attachment added from Website.  Thanks Breiler.

[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/5771587/messages.log)

breiler, 2021-01-05:
That didn't give any clues other than that the visualizer couldn't be properly started. Try closing that window inside of UGS if it is open and restart.

benclewett, 2021-01-05:
I've closed all windows within the app.  Back to a blank screen just
showing menu options and 'No file loaded' in the footer.

The CPU is still 100%.

After restart the CPU again settles at 100%.

If I look at threads rather than processes, the Java shows just one thread
causing the load:

top - 18:22:17 up  1:36,  4 users,  load average: 1.00, 1.08, 0.93
Threads: 240 total,   2 running, 238 sleeping,   0 stopped,   0 zombie
%Cpu(s): 25.3 us,  0.4 sy,  0.0 ni, 74.3 id,  0.0 wa,  0.0 hi,  0.0 si,
 0.0 st
MiB Mem :    873.3 total,    106.5 free,    292.7 used,    474.1 buff/cache
MiB Swap:    100.0 total,    100.0 free,      0.0 used.    517.5 avail Mem

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+
COMMAND
 2494 pi        20   0  453104 187476  22836 R  99.9  21.0  13:35.08
Service Thread

I think this might be of interest:
https://bugs.openjdk.java.net/browse/JDK-8231612

Ben

breiler, 2021-01-05:
Thanks!

If you feel like experimenting you can try installing an older version of java (the newer ones does not work with netbeans).
https://docs.datastax.com/en/jdk-install/doc/jdk-install/installOpenJdkDeb.html

Then download UGS without a bundled Java: https://ugs.jfrog.io/ugs/UGS/v2.0.7/ugs-platform-app.zip

You can verify that it uses the Java 8 if you open the "About" dialog.

benclewett, 2021-01-05:
Sounds like fun.

Before
$ java --version
openjdk 11.0.9.1 2020-11-04
OpenJDK Runtime Environment (build 11.0.9.1+1-post-Raspbian-1deb10u2)
OpenJDK Server VM (build 11.0.9.1+1-post-Raspbian-1deb10u2, mixed mode)

After

$ java -version
openjdk version "1.8.0_212"
OpenJDK Runtime Environment (build 1.8.0_212-8u212-b01-1+rpi1-b01)
OpenJDK Client VM (build 25.212-b01, mixed mode)

I found the --jdkhome switch on your program, and pointed it at 1.8.

The CPU is now on 0.3% as I look at it now.  Problem fixed.

I now have another problem when I open up a file, I get the error:

  No GLDrawableFactory available for profile: GL4bc

I've not connected it to my 3018 yet, as I've blown the PCB and am waiting
for a replacement.

The only thing it is, is a little slow.  I'm using an RPI 3.  I think I may
need an RPI 4.  But that's my problem.

Thanks!

breiler, 2021-01-05:
I'll consider changing the default JRE to 1.8 for RPi, thanks for debugging.

benclewett, 2021-01-05:
Glad to help. If you want me to test any other versions please ask.

breiler, 2021-01-06:
I made a version that bundles an an older JDK, if you have the time I'd appreciate if you could give it a try:
https://drive.google.com/file/d/1sNFKdpGXMmDkd57qSAo43jJszZwltF9K/view?usp=sharing

benclewett, 2021-01-06:
I'm sorry but this still shows the 100% CPU:

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+
COMMAND
 1803 pi        20   0  452908 266792  22748 R  99.7  29.8   5:03.06
Service Thread

This is the /proc/[pid]/cmdline:  (I've installed UGS in
/usr/local/ugsplatform-pi)

/usr/local/ugsplatform-pi/jdk/jdk-13.0.1+9/bin/java-Djdk.home=/usr/local/ugsplatform-pi/jdk/jdk-13.0.1+9-classpath/usr/local/ugsplatform-pi/platform/lib/boot.jar:/usr/local/ugsplatform-pi/platform/lib/org-openide-modules.jar:/usr/local/ugsplatform-pi/platform/lib/org-openide-util.jar:/usr/local/ugsplatform-pi/platform/lib/org-openide-util-lookup.jar:/usr/local/ugsplatform-pi/platform/lib/org-openide-util-ui.jar-Dnetbeans.dirs=/usr/local/ugsplatform-pi/etc:/usr/local/ugsplatform-pi/ide:/usr/local/ugsplatform-pi/bin:/usr/local/ugsplatform-pi/ugsplatform:/usr/local/ugsplatform-pi/extra:/usr/local/ugsplatform-pi/java:/usr/local/ugsplatform-pi/platform:-Dnetbeans.home=/usr/local/ugsplatform-pi/platform-Xms64m-Xverify:none-Dsun.java2d.noddraw=true-Dsun.awt.noerasebackground=true-Dnetbeans.indexing.noFileRefresh=true-XX:+HeapDumpOnOutOfMemoryError-XX:HeapDumpPath=/home/pi/.ugsplatform/2.0.7-SNAPSHOT/dev/var/log/heapdump.hproforg.netbeans.Main--cachedir/home/pi/.ugsplatform/2.0.7-SNAPSHOT/dev/var/cache--userdir/home/pi/.ugsplatform/2.0.7-SNAPSHOT/dev--brandingugsplatformroot@raspberrypi
:/usr/local

Ben

benclewett, 2021-01-06:
That version of Java might have another issue.  My RPi has a mounted Google
Drive:

$ df -T /home/pi/gdrive
Filesystem     Type        1K-blocks    Used Available Use% Mounted on
gdrive:        fuse.rclone  15728640 4250184   2339348  65% /home/pi/gdrive

All other apps can read this drive, accept that Java.  It just shows an
empty drive.  The 1.8 version worked here.

Ben

PR #1506 (2021-01-07): Now using a 32-bit JRE for Windows and JRE8 for raspberry pi.
Changed the default JRE:s for Windows and RaspberryPi.
Fixes #1493 
Fixes #1505
closed 2021-01-07 by breiler