winder/Universal-G-Code-Sender #1507: Visualizer not updating when making changes to Gcode, file not being send when it is updated

jimmys01, 2021-01-07:
**Describe the bug**
When you change/add/remove gcode commands from an opened file the visualizer does not update accordingly.
Also the file is not send to the controller. You have to save the file and the re-open it to mitigate this.

**To reproduce**
Steps to reproduce the behavior:
Open a gcode file
On the editor add commands
the commands are not send to the controller and also are not vizualized

**Expected behavior**
Update the visualizer real time or near real time 
and send the file to the controller each time it has changed and you press the green Send button

**Version**
UGS Platform 2.0.7 windows 10

**Hardware**
GRBL 1.1f ESP32

Windows 10

breiler, 2021-01-07:
I never noticed this, wondering if this could be a Windows issue:
[image]

Could you post the log file, it might have some clues:
https://github.com/winder/Universal-G-Code-Sender/wiki/Configuration#log-files

jimmys01, 2021-01-07:
You are amazing fast :-). In the video you posted the behavior is right. 
[ugs.zip](https://github.com/winder/Universal-G-Code-Sender/files/5783242/ugs.zip)
I am fooling arount a lot with UGS I hope there are not many trash in the logs...

breiler, 2021-01-07:
The logs were filled with errors but I can't find any trace of our code being faulty. This looks suspicious though:
```
	at java.desktop/javax.swing.RepaintManager.paint(Unknown Source)
	at java.desktop/javax.swing.JComponent._paintImmediately(Unknown Source)
	at java.desktop/javax.swing.JComponent.paintImmediately(Unknown Source)
	at java.desktop/javax.swing.RepaintManager$4.run(Unknown Source)
INFO [null]: Last record repeated again.
	at java.base/java.security.AccessController.doPrivileged(Unknown Source)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.desktop/javax.swing.RepaintManager.paintDirtyRegions(Unknown Source)
INFO [null]: Last record repeated again.
```

I am wondering if this could be caused by a antivirus program or you could try running UGS with administrator rights?

jimmys01, 2021-01-07:
[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/5783389/messages.log)
Same behavior with admin rights. No AV except the windows one.
This log is with admin rights. I opened closed and changed the gcode. Hope there is something helpful there

breiler, 2021-01-07:
No errors displayed in this log. I need to test this on a Windows machine to see if I can reproduce the error.

jimmys01, 2021-01-07:
https://user-images.githubusercontent.com/17748323/103934590-e6a6c780-512d-11eb-9be6-d9fb1a776459.mp4
Check the "rows in file" and the visualizer are not updating

jimmys01, 2021-01-10:
any news about it?

breiler, 2021-01-10:
I managed to reproduce the error on Windows 10, but I still don't know why it doesn't work.
I'm currently reworking the file opening code which might fix these issues, but it is still a lot of work left.

breiler, 2021-01-21:
I think I managed to fix this in the latest nightly build. I would appreciate if you could give it a try.

jimmys01, 2021-01-21:
I would like to try it but where to find the build?

breiler, 2021-01-21:
Sorry!

Choose the nightly build from here:
https://github.com/winder/Universal-G-Code-Sender#downloads

jimmys01, 2021-01-21:
Uploading circle.gcode - Universal Gcode Platform (Version 2.0.7-SNAPSHOT _ Jan 21, 2021) 2021-01-21 14-45-55.mp4…
Thanks for the update!!
Still not updating the visualization 
also the rows in file count in the right.

breiler, 2021-01-21:
Really annoying... (I'm cursing at NetBeans and Windows for not working the same way on Linux and Mac)
I did a quick test and it reloaded with a new file but I didn't try to load a file that I previously had loaded. I think that my latest change will work. I'll let you know when it's merged and built.

breiler, 2021-01-22:
Its merged and built, available in the latest nightly build

jimmys01, 2021-01-22:
It will update the visualization when I go to File->save but not in real time.
For my work case this is fine but it is not the same behavior  as you show in your first video

breiler, 2021-01-22:
I was pressing the shortcut for saving so that it could update the visualiser. There was never a real time update.

As a developer I took this behaviour for granted. I.e. in all IDE's it requires the file to be saved for it to compile into a program. If anyone thinks that this behaviour should change, create a new issue so that we could discuss it separately.

Thanks for the issue report and testing @jimmys01!

closed 2021-01-22 by breiler

jimmys01, 2021-01-22:
Ctrl+s is not saving the file by the way... Am i missing something? At the File submenu its is asking for Meta+S instead of Ctrl+S. Perhaps this is a new windows bug thing?