winder/Universal-G-Code-Sender #1517: program not starting properly

faisalkamalia, 2021-01-19:
Program loads but at welcome screen it stuck and seconds after closes.
[image]
after this screen it not goes to next step.

windows 8.1

breiler, 2021-01-19:
Could you attach a log file:
https://github.com/winder/Universal-G-Code-Sender/wiki/Configuration#log-files

palestinaikos, 2021-01-23:
Not sure if related, but I also can't get it started properly. 
[image]
The log was saved as txt so that github accepts it.
Linux pef7 5.4.0-62-generic #70-Ubuntu SMP Tue Jan 12 12:45:47 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
[messages.log.2.txt](https://github.com/winder/Universal-G-Code-Sender/files/5861244/messages.log.2.txt)

I used it without problems on Linux Mint 19 Cinnamon, but after upgrading to v20, I can't get it working again. The same problems arise with stable version as well as with the current nightly build.

Can I provide more info to help with diagnostics? And if so, how? (Sorry, Java is way out of my league :D )

breiler, 2021-01-24:
@palestinaikos from the log this seems like another problem. It complains about a GL-related problem:
`Caused by: com.jogamp.opengl.GLException: Caught GLException: Profile GL4bc is not available on X11GraphicsDevice``

Does the program `glxgears`work? What kind of graphics card are you using? 

Similar problems have been reported here:
https://github.com/winder/Universal-G-Code-Sender/issues/1391

faisalkamalia, 2021-01-24:
> Could you attach a log file:
> https://github.com/winder/Universal-G-Code-Sender/wiki/Configuration#log-files

i attached log file please check sir. thanks

[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/5861745/messages.log)

palestinaikos, 2021-01-24:
> @palestinaikos from the log this seems like another problem. It complains about a GL-related problem:
> `Caused by: com.jogamp.opengl.GLException: Caught GLException: Profile GL4bc is not available on X11GraphicsDevice``
> 
> Does the program `glxgears`work? What kind of graphics card are you using?

`glxgears` works. `308 frames in 5.0 seconds = 61.441 FPS`

Graphics card: 
`lspci | grep VGA`:
`00:02.0 VGA compatible controller: Intel Corporation HD Graphics 500 (rev 0b)`

I also tried other versions (stable & nightly builds, Linux & All platforms), but only succeeded with running UGS Classic. 

> Similar problems have been reported here:
> #1391

You are correct, my screenshot above looks exactly like <https://github.com/winder/Universal-G-Code-Sender/issues/1391#issuecomment-629631983>. But after following the advice in the #1391 thread, I removed ~/.ugs and ~/.ugsplatform, I managed to replicate the same behaviour as <https://github.com/winder/Universal-G-Code-Sender/issues/1517#issue-788719228> for the nightly build. Stable version seemed to work at first, but soon proved to be glitchy beyond usability.

[image]
When running `sudo ./ugsplatform`, the dark theme is no longer used, but behaviour doesen't change much.

[image]

Latest log: [messages.log.2.txt](https://github.com/winder/Universal-G-Code-Sender/files/5862020/messages.log.2.txt)

breiler, 2021-01-24:
The log file is full of GL-specific errors, it seems to be resolved with the 2.4.0-release of Jogamp which is not available yet:
http://forum.jogamp.org/Crash-after-latest-mesa-driver-update-from-ppa-oibaf-td4040300.html

sproutecks2125, 2021-01-31:
Wanted to post a comment about this not working on Windows 10 as well. I can't seem to even find the log file as indicated in the docs (i.e. no "ugsplatform") file in C:\..\AppData\

I tried to read through some of the links provided in here by other but did not get to a solution. 
Screenshot attached showing "jogamp_exe..." file and where it is hanging the program.

Look forward to a fix as I am really excited to try your software with my xcarve.

Thank you for your work on this project.
[image]

faisalkamalia, 2021-02-01:
> > Could you attach a log file:
> > https://github.com/winder/Universal-G-Code-Sender/wiki/Configuration#log-files
> 
> i attached log file please check sir. thanks
> 
> [messages.log](https://github.com/winder/Universal-G-Code-Sender/files/5861745/messages.log)

breiler, 2021-02-01:
@faisalkamalia did you solve this, and if so how?

faisalkamalia, 2021-02-01:
no I did not solved this, mine problem is that please check the above comment contains log file.

breiler, 2021-02-01:
@faisalkamalia Your log file didn't give any clues. Do you have an Antivirus program that might block things?

faisalkamalia, 2021-02-01:
no i did not have a antivirus. leave it. thanks i will do it my self.

closed 2021-02-01 by faisalkamalia