winder/Universal-G-Code-Sender #1521: Macros added to toolbox grey'ed out each time I restart.

chrisla23, 2021-01-21:
**Describe the bug**
I have created some macros. If I add them to the "toolbox" they work fine, but then the next time I restart they are greyed out and unusable. If I go through settings, remove them and re-add them, they are normal again and work. 

**To reproduce**
Steps to reproduce the behavior:

- Machine -> Edit Macros
-  Click +Add 
- Provide a name and valid gcode
- Click Try, confirm the operation
-  Click OK
- Tools -> Options -> UGS -> toolbox
-  Find macro on the right panel, double click to add to the toolbox
-  Click OK
-  Confirm operation in the toolbox
- Restart UGS
- Do normal reconnection, homing, etc
- Confirm CNC is responsive and normal
- Macro appears in the toolbox but is unusable
- Confirm macro still works via Machine -> Macros -> Macro name
- Go through steps to remove and re-add, it works again.

**Expected behavior**
I expect the macro to be usable after a restart and startup procedures. 

**Screenshots**
Screenshot attached.

[image]

**Version**
Product Version: Universal Gcode Platform 20201202 
Java: 13.0.1; OpenJDK 64-Bit Server VM 13.0.1+9 
Runtime: OpenJDK Runtime Environment 13.0.1+9 
System: Windows 7 version 6.1 running on amd64; Cp1252; en_US (ugsplatform) 
User directory: C:\Users\Administrator\.ugsplatform\2.0.7-SNAPSHOT\dev 
Cache directory: C:\Users\Administrator\.ugsplatform\2.0.7-SNAPSHOT\dev\var\cache 

**Hardware**
grbl / Bobscnc

**Operating system (please complete the following information):**
 Windows 7

**Additional context**
None

Thanks,

-Chris

breiler, 2021-01-22:
Thanks for the detailed report!

I found the issue and a fix is on its way.

riaanmarx, 2022-01-06:
Greetings everyone.
I am (still) experiencing this issue on the following platform:
Product Version: Universal Gcode Platform 20211124
Java: 13.0.1; OpenJDK Client VM 13.0.1+9
Runtime: OpenJDK Runtime Environment 13.0.1+9
System: Windows 10 version 10.0 running on x86; Cp1252; en_US (ugsplatform)

The grayed-out/disabled buttons are all custom macros.
[image]
(this screenshot was taken after I restarted UGS, connected to GRBL and homed the machine)

Can anyone else please confirm if they also experience this?
Thank you and best regards,
Riaan

breiler, 2022-01-07:
I could reproduce this, I'm will look into it!

breiler, 2022-01-20:
This should is resolved in UGS 2.0.10

PR #1522 (2021-01-22): Fixed problem with inactive macro actions, now listens to UGS events
Fixes #1521
closed 2022-01-20 by breiler