winder/Universal-G-Code-Sender #1523: Text Mode (#ej=0) breaks position readout until return to json ($ej=1) and status update (?)

Mgilbride, 2021-01-25:
First of all, first time I have touched UGS in a long time. First time using platform. So thankful for g2 support. Nice job!! Thank you.

I'm running g2core master branch (100.26) on a due.

I noticed that if you switch to text mode to review settings, status, position reports and then move (ex. g0 x1000) the DRO does not update. Switch back to json mode ($ej=1) and check status (?) and position will  update again. if you move while in json mode ($ej=1) then the DRO works as intended. Maybe this is by design. But, am guessing not. I have a work around by using a macro that sets json mode at start and end of settings, status, position queries. Maybe it should be handled from the console to avoid confusion. 

Thanks again.

breiler, 2021-01-26:
Hi @Mgilbride, this is by design. The JSON-mode is better suited for Machine-to-Machine communication and we have only implemented that part of the protocol. You could try to use the Firmware settings dialog for reviewing your settings, its not perfect but I have found it useful. https://github.com/winder/Universal-G-Code-Sender/wiki/Configuration#firmware-settings

The 100.26 branch did contain a bug that does not report the machine status properly and I would recommend using version 101.03 or later.

Mgilbride, 2021-01-26:
Got it. Thanks. I'll try using 101.03. Tried edge and was having too many issues.

closed 2021-02-04 by breiler