winder/Universal-G-Code-Sender #1527: Error while processing response <<Idle|MPs:0.000,0.000,0.000|FS:0,0>>

Snow4DV, 2021-01-31:
**Describe the bug**
I've just installed clean grbl v1.1h, latest version of software (didn't modify any settings) and got a rainbow of parsing errors:
`Error while processing response <<IdleMPos:0.000,0.000,0.000|FS:0,0>>
[Error] Error while processing response <<Idle|MPos:0.000,0.000,0.000|FS:0,|WCO:0.000,0.000,0.000>>: Index 1 out of bounds for length 1
[Error] Error while processing response <<Idle|Mos:0.000,0.000,0.000|FS:0,0>>
[Error] Error while processing response <<IdleMPos:0.000,0.000,0.000|FS:0,0>>
[Error] Error while processing response <<Idle|MPo:0.000,0.000,0.000|FS:0,0>>
[Error] Error while processing response <<Idle|MPos:0.000,0.000,0.000|FS:0,>>: Index 1 out of bounds for length 1
[Error] Error while processing response <<Idle|Pos:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000>>
[Error] Error while processing response <<dle|MPo:0.000,0.000,0.000|FS:0,0>>
[Error] Error while processing response <<IdleMPos:0.000,0.000,0.000|FS:0,0>>
[Error] Error while processing response <<Idle|MPos0.000,0.000,0.000|FS:0,0>>
<Idle|MPos:0.000,0000,0.000|FS:0,0|WCO:0.000,0.000,0.000
[Error] Error while processing response <<Idle|MPos:0.000,0.0000.000|FS:0,0|Ov:100,100,100>>
Idle|MPos:0.000,0.000,0.000FS:0,0>
[Error] Error while processing response <<Idle|MPos:0.000,0.000,0.000|FS:00>>: Index 1 out of bounds for length 1
[Error] Error while processing response <<Idle|MPos:0.000,0.000,0.000|FS:0,>>: Index 1 out of bounds for length 1
[Error] Error while processing response <<Idle|MPos:0.000,0.0000.000|FS:0,0>>
[Error] Error while processing response <<Idle|MPos:0.000,0.0000.000|FS:0,0>>
[Error] Error while processing response <<Idle|MPo:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000>>
Idle|MPos:0.000,0.000,0.000|FS:0,0>
[Error] Error while processing response <<Idle|Mos:0.000,0.000,0.000|FS:0,0>>
[Error] Error while processing response <<Idle|Mos:0.000,0.000,0.000|FS:0,0>>
[Error] Error while processing response <<Idle|MPos:.000,0.000,0.000|FS:0,|Ov:100,100,100>>: Index 1 out of bounds for length 1
<Idle|MPos:0.000,0.000,0.000|FS:0,0`
**To reproduce**
Steps to reproduce the behavior:
1. Upload latest grbl release to arduino uno, v1.1h
2. Install ugs platform
3. Open it and connect to the machine

**Expected behavior**
Was expecting not to see any of these :)

**Screenshots**
[Screenshot_54](https://user-images.githubusercontent.com/32912696/106380951-ac47e780-63c6-11eb-94ed-8d532cc7759e.png)

**Version**
 UGS Platform 2.0.7 snapshot

**Hardware**
GRBL 1.1h

**Operating system (please complete the following information):**
Win 10 1909

breiler, 2021-01-31:
It looks like you don't have a stable connection to the controller. The messages from the controller aren't complete, for instance:  `<IdleMPos:0.000,0.000,0.000|FS:0,0>` should have been `<Idle|MPos:0.000,0.000,0.000|FS:0,0>`

Try it with another USB-cable. What kind of Arduino are you using? 
Is it a cheap China variant or a genuine Arduino?

Snow4DV, 2021-02-01:
Thanks! Yeah, it's the cheap one. But that's strange how it didn't affect other things in the grbl. Gonna search for the answers in their repo

closed 2021-02-01 by Snow4DV

breiler, 2021-02-01:
@Snow4DV I'm not sure that it is the problem, I have not experienced this myself. But we have seen similar quality related problems like this before. I would try with another USB cable first.

Snow4DV, 2021-02-01:
> @Snow4DV I'm not sure that it is the problem, I have not experienced this myself. But we have seen similar quality related problems like this before. I would try with another USB cable first.

I've tried three cables, one of them is pretty good and all resulted to the same behaviour - so probably that's about arduino or its software)