winder/Universal-G-Code-Sender #1528: when moving tool from Zero setting, it doesn't go up and drags on the surface

lebi55, 2021-02-01:
**Describe the bug**
A clear and concise description of what the bug is.

**To reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Version**
[e.g. UGS Platform 2.0.6, UGS Classic 1.0.9]

**Hardware**
[e.g. GRBL 1.1f, TinyG,  SmoothieBoard]

**Operating system (please complete the following information):**
 [e.g. Windows 10, MacOSX, Ubuntu 20.04 ]

**Additional context**
Add any other context about the problem here.

breiler, 2021-02-01:
I have no idea what you mean, you need to elaborate.

lebi55, 2021-02-01:
Hi
Thanks for the reply
To do engraving I bring the tool to the origin and bring the tool down to touch the top surface and I set zero on x y and z
When I start the program, the tool should go up and then move fast to the first position
In my case the tool doesn’t go up bit remain down at zero on the surface and thereby drag the tool, making marks
At the GO command there is no Z plus command
Please help
Regards
Lebi

breiler, 2021-02-01:
What software do you use to generate the Gcode?
Do you have a sample file you could share?

lebi55, 2021-02-01:
Hi
I am using Fusion 360(free version for hobbyist)
[cid:5C11256E-2491-44EE-BD60-A5C2C3B52D4C-L0-001]

breiler, 2021-02-01:
You should be able to find tutorials for setting clearance heights on Youtube. Maybe someone else here know how to do this (feel free to post a reply) but I'm closing this issue as it is not related to UGS.

closed 2021-02-01 by breiler

lebi55, 2021-02-01:
Hi
thanks for replying
So it is the program’s fault that there is no Z plus command?
When I edited the program to z0.1< it was working ok
Regards
lebi