winder/Universal-G-Code-Sender #1536: UGS 2.0.7 unresponsive on Win10

esawdust, 2021-02-13:
**Describe the bug**
When launching UGS platform on Windows 10, the application comes up but has a popup dialog stating it's a nightly build.   The checkbox on the dialog and the OK button to dismiss the dialog are both unresponsive.   This makes the entire application unusable because the dialog seems to modal and I can't get by it and it is impossible to interact with the dialog.

UGS classic also pops the same type of nightly build dialog, but I am able to click by it and dismiss it. 

UGS classic works fine, UGS platform does not - same machine, same windows login session.

I've tested 2.0.7 platform on OS X Catalina 10.15.7 as well - no problem.  2.0.7 platform also on Ubuntu Xenial 16.04 and was able to advance past the dialog as well.   

**To reproduce**
Steps to reproduce the behavior:
1. Launch UGS platform 2.0.7 application on Windows 10
2. Nightly build dialog comes up but cannot be dismissed - interaction with dialog checkbox and OK dismissal is hung and dialog is completely unresponsive.

**Expected behavior**
Interact with and be able to dismiss the nightly build dialog without it hanging the application

**Screenshots**
Nothing special here - just the nightly build modal dialog on top of the application.

**Version**
UGS Platform 2.0.7

**Hardware**
HP laptop

**Operating system (please complete the following information):**
Windows 10

**Additional context**
See discription

breiler, 2021-02-13:
Thanks, can we have your log file
https://github.com/winder/Universal-G-Code-Sender/wiki/Configuration#log-files

rlwoodjr, 2021-02-16:
I also am having an issue with the firmware loading on the nightly build.  The same firmware loads on the latest stable and on the nightly classic build.  

The Console window displays up to $21, ten stops.

[messages.log](https://github.com/winder/Universal-G-Code-Sender/files/5991496/messages.log)

breiler, 2021-02-17:
Thanks @rlwoodjr, I think you've experienced an other error caused by the new ABC-axises which we need to have a look at:
```
WARNING [com.willwinder.universalgcodesender.AbstractCommunicator]: Couldn't send event
java.lang.IllegalArgumentException: Tried to get a-axis which is not set
	at com.willwinder.universalgcodesender.model.PartialPosition.getA(PartialPosition.java:143)
	at com.willwinder.universalgcodesender.model.PartialPosition$Builder.copy(PartialPosition.java:333)
	at com.willwinder.ugs.nbm.visualizer.renderables.MachineBoundries.onUGSEvent(MachineBoundries.java:70)
	at com.willwinder.universalgcodesender.model.GUIBackend.lambda$sendUGSEvent$3(GUIBackend.java:894)
	at java.base/java.util.ArrayList.forEach(Unknown Source)
	at com.willwinder.universalgcodesender.model.GUIBackend.sendUGSEvent(GUIBackend.java:894)
	at com.willwinder.universalgcodesender.model.GUIBackend.onUpdatedFirmwareSetting(GUIBackend.java:917)
	at com.willwinder.universalgcodesender.firmware.grbl.GrblFirmwareSettingsCommunicatorListener.lambda$rawResponseListener$1(GrblFirmwareSettingsCommunicatorListener.java:237)
	at java.base/java.lang.Iterable.forEach(Unknown Source)
	at java.base/java.util.Collections$SynchronizedCollection.forEach(Unknown Source)
	at com.willwinder.universalgcodesender.firmware.grbl.GrblFirmwareSettingsCommunicatorListener.rawResponseListener(GrblFirmwareSettingsCommunicatorListener.java:237)
	at com.willwinder.universalgcodesender.firmware.grbl.GrblFirmwareSettings.rawResponseListener(GrblFirmwareSettings.java:417)
	at com.willwinder.universalgcodesender.AbstractCommunicator.sendEventToListeners(AbstractCommunicator.java:172)
[catch] at com.willwinder.universalgcodesender.AbstractCommunicator.lambda$new$0(AbstractCommunicator.java:194)
	at java.base/java.lang.Thread.run(Unknown Source)
```

AndValderrama, 2021-02-28:
i´m having exactly the same problem, does anyone have found the solution?

Bosque00, 2021-03-22:
I am also having the issue with not being able to dismiss the nightly build dialogue box. Was anyone able to resolve this?
[image]

Bosque00, 2021-03-23:
I was able to get UGS platform 64 to launch by using the following steps: 
1) Restart your computer
2) Go to /home/user/ugs on windows and delete all files
3) Go to /home/user/ugsplatform on windosw and delete all files
4) Launch ugsplatform64
5) press the X in the upper right to close the dialogue box as soon as possible

I am not sure that steps 2 and 3 both need to be done. 

Also I noticed that after I do this I cant launch UGSplatform 64 again if I close it using the X in the upper right of the GUI. But I can if I use File>Quit it seems that I can reopen ugsplatform64. 

This leads me to think that something is not being closed properly when you use the X in the upper right of the GUI.

Bosque00, 2021-03-23:
I just tested the above without steps 2 and 3 and did get ugsplatform64 to launch.