winder/Universal-G-Code-Sender #1537: Decrease the Joystick step size limit to 0.001 for inch mode

htrantx, 2021-02-13:
when using joystick feature the step size for the machine is limited to 0.01, which is fine for mm but not for inch mode.

Changing the step size with Machine/Jog/Stepsize has no effect on the joystick action

Please change the joystick limit to 0.001 for inch mode

breiler, 2021-02-14:
This is most likley caused by this error:
https://groups.google.com/g/universal-gcode-sender/c/q5wdEQSZZTU

There is a possible fix on its way:
https://github.com/winder/Universal-G-Code-Sender/pull/1538

breiler, 2021-02-16:
Try the latest nightly build, I think that I have solved the problem with the dynamic actions that does not load properly on some platforms.

It should look something like this:
[image]

htrantx, 2021-02-17:
Thanks Joecim,

It's fixed!

closed 2021-02-17 by htrantx