winder/Universal-G-Code-Sender #1541: Crashes on start

G-Pereira, 2021-02-18:
**Describe the bug**
The app runs and when the main window appears it crashes after 2 seconds.

**To reproduce**
Steps to reproduce the behavior:
1. clone repo
2. run `./run_platform.sh`

**Expected behavior**
I would expect that the window stays open so I can use the program :)

**Screenshots**
[image]

**Version**
from master (hash `50e4bc471c1de429324a2e4e076411ea14bf74b6`)

**Hardware**
(not applicable)

**Operating system (please complete the following information):**
Fedora 33 (Java 11)

breiler, 2021-02-18:
What happens if you run the built stable or nightly version from here:
https://github.com/winder/Universal-G-Code-Sender#downloads

There has been problems getting it to run on certain JVM:s, we are bundling the linux builds with JRE 13 which seem to have worked on several Linux variants.

G-Pereira, 2021-02-18:
> What happens if you run the built stable or nightly version from here:
> https://github.com/winder/Universal-G-Code-Sender#downloads

Under nightly I should choose "All platforms", right?

> There has been problems getting it to run on certain JVM:s, we are bundling the linux builds with JRE 13 which seem to have worked on several Linux variants.

I see, that's useful information. How can I set which version to use for compilation? Is `update-alternatives --config java` enough?

breiler, 2021-02-18:
No I was thinking that you should try the Linux build to make sure that the bundled JVM works as intended, it is using Java version 13.0.1 from here: https://adoptopenjdk.net/

If you would like to build it yourself  the `update-alternatives --config java` should work, but I do not know which versions you should try. On MacOSX I'm running Java 8 and Java 13.0.1.

Do you want to build it because you are developing or just for running it?

G-Pereira, 2021-02-19:
When I use a pre-compiled version I always get this:
[image]

I am trying to use my CNC and it seems that the Linux support of existing gcode loaders is very poor. I haven't found a tool that works out of the box. I got a specific Candle version running but it also has some issues in the UI so I wanted to try UGS.

lamikr, 2021-04-20:
I was getting similar type of "bad ui" errors on Linux with all of the latest versions. (206, 207) Latest version I got pretty much working was a non-platform version of 2.0.3. I believe that the problem is with the jogl 2.3.2 version that does not work anymore well with the newest Linux distro versions. 

These older versions have worked ok for me to send gcode to grbl 1.1h based  laser and cnc. I might have needed to change the app to use the JSSC library instead of java's own ones. (Change can be done in ugs settings if I remember correctly)

But I have also now got the latest 2.0.7 platform UI showing up correctly by doing a local Linux build. It's a little hackish because gluegen 2.4.0 and jogl 2.4.0 has not been released to maven repositories yet.
I have now cloned the source code and got things pretty much working by updating the gluegen and jogl to unreleased 2.4.0 version that I have also build by myself and then changed the 2.3.2 version references to 2.4.0 in pom.xml. My solution is little hackish because there is no jogl 2.4.0 jars on maven and therefore I needed to manually package gluegen and jogl 2.4.0 versions to my loical maven repo directories:

    .m2/repository/org/jogamp/gluegen/gluegen-rt-main/2.4.0/
    .m2/repository/org/jogamp/gluegen/gluegen-rt/2.4.0/
    .m2/repository/org/jogamp/jogl/jogl-all-main/2.4.0/
    .m2/repository/org/jogamp/jogl/jogl-all/2.4.0/

I also needed to force the maven to prefer these local 2.4.0 versions instead of trying to download them from the internet by creating a settings.xml with single line: 
<offline>true</offline>

I tried to upload the zip file which would contain this local maven repo and changes but for some reason it is failing.

breiler, 2022-11-04:
Please try the latest nightly build, I have updated JOGL to 2.4.0 and I don't have these issues on my machine anymore.

breiler, 2023-08-21:
Closing as this should have been resolved with the newer version of the the JOGL-library.