winder/Universal-G-Code-Sender #1542: NSInternalInconsistencyException - NSWindow drag regions should only be invalidated on the Main Thread!

dvsingletary, 2021-02-18:
When trying to build/run UGS platform by cloning the repository and using the run_platform.sh script, I get the following exception:
2021-02-18 10:09:57.620 java[32125:1062505] *** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'NSWindow drag regions should only be invalidated on the Main Thread!'
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fff206496af __exceptionPreprocess + 242
	1   libobjc.A.dylib                     0x00007fff203813c9 objc_exception_throw + 48
	2   CoreFoundation                      0x00007fff20671a9a -[NSException raise] + 9
	3   AppKit                              0x00007fff22e18460 -[NSWindow(NSWindow_Theme) _postWindowNeedsToResetDragMarginsUnlessPostingDisabled] + 321
	4   AppKit                              0x00007fff22e03121 -[NSWindow _initContent:styleMask:backing:defer:contentView:] + 1375
	5   AppKit                              0x00007fff22e02bbb -[NSWindow initWithContentRect:styleMask:backing:defer:] + 42
	6   libnativewindow_macosx.jnilib       0x00000001437973fe Java_jogamp_nativewindow_macosx_OSXUtil_CreateNSWindow0 + 398
	7   ???                                 0x0000000123513c90 0x0 + 4887493776
)
libc++abi.dylib: terminating with uncaught exception of type NSException

To Reproduce:
1) clone the repository
2) run_platform.sh (builds and runs with the latest installed Java) - will crash with the above message.
3) will also crash in the same way if you run ugs-platform/application/target/ugsplatform/bin/ugsplatform with --jdkhome pointed to any of the the following JDKs:
adoptopenjdk-11.jdk
jdk-13.0.1.jdk <-- I found this especially odd, since this is the version that the compiled binary is shipped with!
jdk-13.0.2.jdk
jdk-15.0.2.jdk
openjdk-13.0.1.jdk
jdk1.8.0_281.jdk

OF NOTE is that everything works OK if I use the following older JDK:
jdk1.8.0_241.jdk

This is on MacOS 11.2.1

I am would love to figure out what's going on here.  Willing to produce any additional information you need to nail this down.  I've seen this issue reported by others here but couldn't find any definitive solution other than running against an old JDK 1.8.0_281.

breiler, 2021-02-18:
Welcome back @dvsingletary!

I can build using adoptopenjdk-13.0.1 on MacOS 10.12.6 (I know, it's old but don't have space to upgrade 🤦 )
This issue will be fixed once we get the new release of JOGL 2.4.1 (I have tested this with a snapshot build), until then we are stuck with older JDKs...

dvsingletary, 2021-02-18:
Thanks!

It wasn't clear to me earlier if this was a known issue.  Glad that you're aware and that using the older JDK 8u241 is an acceptable solution.

closed 2021-02-22 by breiler