winder/Universal-G-Code-Sender #1544: Implementing custom features on the Raspberry PI

krupis, 2021-02-23:
Hello, I am working on a project and I have grblHAL running on ESP32 and Raspberry pi running G code sender.

I want to make Raspberry PI communicate with some external pick and place machinery, that is programmed to place to items inside the CNC machine. 

So initially, I would load the G code in the machine, and start the pick and place machinery.
The machinery is checking various sensors/states and sucesfully places an item that needs milling inside a holder and is ready to start the G code. 

Instead of manually pressing the button, I want to automate the process and get pick and place machinery to send a signal to raspberry PI. After raspberry PI receives that signal, it will start the g code and mill the part. After milling is completed, the CNC goes into idle state, and waits for another item to be placed.

Also, I would like Raspberry PI to be able to send a message or toggle a pin when its doing probe XYZ test. If a probe test for whatever reason fails ( the drill bit is broken or whatever other reason) , the G code sender should execute some G code which would move the spindle towards an operator so he can inspect whats wrong.  Also, a message will be sent to the external machinery which will be programmed to display a message "PROBE TEST FAILED" when it reiceves some message from the raspberry PI.

Can someone advice how can this be achieved! Appreciate any help

breiler, 2021-02-23:
I think that you could achieve most  of this using the pendant API in UGS. There isn't much documentation on it (I was never able to get Swagger be included in a netbeans module) but the code is pretty self explanatory:
https://github.com/winder/Universal-G-Code-Sender/tree/master/ugs-pendant/src/main/java/com/willwinder/universalgcodesender/pendantui/v1/resources

There isn't any push capabilities in the API such as websockets so you would have to poll the machine status. Let us know if there are any API methods missing. For instance I don't know if you would be able to get the probe status using the API.

krupis, 2021-02-23:
Hello. Thanks for the answer. Can you clarify regarding few things that you mentioned?

I do not think that I need to know the machine status ? The external pick and place machine is  going to send a Raspberry PI a command if everything is ok and ready to start g code, if something is not ready, the command to Raspberry PI will not be send therefore it will remain IDLE.

Also, during a probe, I do not need to know the probe status, if something is wrong , I just display message on my machinery "Something is wrong". I do not need to know what is wrong exactly.

The main concern is how to start the g code after receiving an external command and how to output a signal from Raspberry PI to my external pick and place machine.

The skeleton code:

"External machinery checks everything"
If(READY)
   start_gcode();
else
   do_nothing();

breiler, 2021-02-23:
Use the same HTTP/RPC API that the web pendant uses:
[image]

So your `start_gcode()` could be done using a script with curl:

Uploading a file:
```curl -v -F file=@my-gcode-file.gcode http://localhost:8080/api/v1/files/uploadAndOpen```

Running it:
```curl -v -X POST http://localhost:8080/api/v1/files/send```

krupis, 2021-02-24:
Thanks for the reply. What if I do not use network on my machines? Do these commands require the raspberry pi to be connected to a internet network?

breiler, 2021-02-24:
No it should not require an internet connection, but the easiest way to make sure is to fire up UGS and go to the web pendant using `http://localhost:8080/`. Just make sure the web pendant is activated to start upon startup in the UGS settings.

Now that I think of it, there is a headless mode of UGS which starts the program in a terminal without a GUI. This would allow you to either start a file transfer using the command line, or you can start it in a daemon mode which would allow you to control the CNC using the same Pendant API. I would prefer the API as this will keep a live connection to CNC machine. The documentation for it is a bit outdated and you might want to try this in a step two: https://github.com/winder/Universal-G-Code-Sender/tree/master/ugs-cli

closed 2021-06-14 by breiler